betaflight/betaflight #3934: ESC Telemetrie Current Sensing way off

BavariaAirview, 2017-08-23:
Hi, 

for this i´m not sure if this is an ESC related Thing or if you could change Firmware on FC.

Using Matek F405 with BF3.2 RC3 and Wraith32 ESC With BLHELI32  Rev32.1.0.1
Voltage Monitoring is via ADC on FC. 
But Current monitoring I use ESC telemetry. It does work as it should but is way off. 
I tested couple Batterys and found out this:  On OSD consumtion was 1000mAh - real via Charger 1300mAh    on 2 min flights 5S 
OSD: 500 - Charger: 820
I would like to have correction Value for ESC Current sensing. But I didn´t found any.

AirBreak69, 2017-08-24:
This topic should be posted in RCgroups (https://rcgroups.com/forums/showthread.php?t=2464844).
But i want to spread some general ideas about "measuring". In addition you may rise a "feature request" (see last chapter).

There are a lot of sources for error in current/capacity measurement:

1. The charger itself: ±2...5%
2. Thermal and chemical losses between charging and discharging depending on current level: +5..+10% charge/discharge ratio
3. Improper scaling. In most cases the firmware in ESCs/FCs is generic and has to run on clones of clones of original designs too. Small design changes in the current sensor / amplifier / ADC reference voltage path may be not tracked.
4. Dead area: Due the primitive/limited electronics (single low voltage supply, non rail-to-rail amplifiers) low currents aren't measured correctly. Example: Real 0.0A may read as 0.5A while 1.5A read as 0.6A but 2.0A read as 1.1A.

If you want to be precise the task is to measure the characteristic curve in comparision to a good external current meter. This may be real difficult and dangerous as it had to be done with a capitvated quad on a lab bench. I did that while leaving the room and monitoring the OSD and amp meter with a video cam!

Joshua Bardwell describes it very (Bard)well. Be aware of the upside down mounted props!
https://www.youtube.com/watch?v=mfM4EstS-Gs

In case of my DYSF4PRO FC i got the following (rounded) CLI parameters:
set ibata_scale = 500 (default = 400)
set ibata_offset = -75 (default = 0)
That leaded to a scaling of 5/4th of the generic defaults (70A max. measurement range instead of 150A specified).
I had to introduce a strange offset: The board/OSD simply starts reacting above real 1.8Amps.
This leads to an unrealistic behaviour in unarmed state where the quad may draw around 0.5Amps but the drawn capacity rises to fast. But during flight you are far above this "triple point" and capacity will be quite correc if you do not stay unarmed for longer periods.

In your case it seems to be very simular: With a 5/4th relative scaling and some offset you may get near the truth.

Unfortunately BF seem not to provide additional scaling/offset for the ESC currents in current.c.
void currentMeterESCRefresh(int32_t lastUpdateAt)
{
...
    escSensorData_t *escData = getEscSensorData(ESC_SENSOR_COMBINED);
    if (escData->dataAge <= ESC_BATTERY_AGE_MAX) {
        currentMeterESCState.amperage = escData->current;
        currentMeterESCState.mAhDrawn = escData->consumption;
    } else {
        currentMeterESCState.amperage = 0;
        currentMeterESCState.mAhDrawn = 0;
    }
} 
If any of the developers aggree you could change the title to "FR: scale/offset for ESC current sensing".

mikeller, 2017-08-25:
Yes, adding calibration for ESC voltage / current data is definitely something that should go in.

Until now, ESC telemetry was only supported with KISS 24A ESCs, and with those, the readings seemed to be quite accurate (suggesting some form of in-factory calibration being done on the ESC). Now that any blheli32 ESC can support telemetry, and with cheaper models probably using high tolerance shunts and no calibration, there is a use case for calibration on the ESC.

kicekilla, 2017-11-27:
This would be amazing, and is needed in my opinion. Ori32 4in1 also has incorrect current usage. and not on the conservative side either.. Currently I have audio readout from taranis tlm and have an offset on the tlm fuel sensor... but this is not the best way to do this and OSD still shows incorrect current obviously.

DieHertz, 2017-11-27:
I think this should be a request to blheli 32 developer

kicekilla, 2017-11-27:
I agree thats the source of the problem(or hardware designer) but the same goes for onboard ADC sensors on FC's and we have a way to tune them...

DieHertz, 2017-11-27:
ADC is an inalienable part of MCU, while in case with ESC sensor ADC is on the ESC, not on the FC. Still if ESC sensor is viewed as a sensor type along with ADC and virtual, there's no harm in adding scale and offset handling for it.
I just suggest sending the request to blheli as well

mikeller, 2018-06-03:
@BavariaAirview: This has been addressed by the addition of scaling to BLHeli_32 (where it belongs), and the addition of #6004 that allows for the compensation for the (constant) consumption of non-ESC components. I think this can be closed.

closed 2018-07-07 by mikeller