betaflight/betaflight #3937: Arming Sequence: If quad is plugged in with a ARMED tx active - it will arm immediately. And vice-versa.

EggsBenedict, 2017-08-23:
In previous versions of Betaflight, either scenario below required you to disarm before you successfully arm:

1. Quad was plugged in and your turned on a TX with the arm switch active.
2. TX with arm switch already active, but quad not plugged in - and you plug in the quad.

Both of these scenarios in 3.2 result in the quad instantly arming and spinning props (if that's enabled).

While neither scenario should really happen (your arm switch should be off) - this does seem like a feature that should be carried over.

Perhaps I have something setup wrong though?

I did search around if this had already been reported but was unable to turn anything up. Hopefully this isn't a duplicate.

```
CLI

# diff
# Betaflight / OMNIBUS (OMNI) 3.2.0 Aug  6 2017 / 13:33:02 (981df1d54)

feature -RX_PPM
feature RX_SERIAL
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER
serial 2 64 115200 57600 0 115200
aux 0 0 0 1650 2100
aux 2 1 3 900 1175
aux 3 2 3 1300 1700
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set baro_hardware = NONE
set rssi_channel = 9
set serialrx_provider = SBUS
set dshot_idle_value = 500
set motor_pwm_protocol = DSHOT600
set align_board_yaw = 90
set bat_capacity = 850
set ibata_scale = 740
set yaw_motors_reversed = ON
set pid_process_denom = 2
set osd_cap_alarm = 550
set osd_vbat_pos = 44
set osd_rssi_pos = 2093
set osd_tim_1_pos = 2423
set osd_tim_2_pos = 33
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_crosshairs = 200
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 2086
set osd_mah_drawn_pos = 2098
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 0
set osd_pid_pitch_pos = 0
set osd_pid_yaw_pos = 0
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 2401
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_disarmed_pos = 2251
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_bbox = OFF
set osd_stat_endbatt = ON
set osd_stat_bb_no = OFF
set osd_stat_tim_2 = OFF
set vcd_video_system = 2
profile 0

set dterm_lowpass_type = PT1
set setpoint_relax_ratio = 40
set dterm_setpoint_weight = 108
set p_pitch = 54
set d_pitch = 8
set p_roll = 36
set d_roll = 12
rateprofile 0

set roll_srate = 72
set pitch_srate = 72
set yaw_srate = 72
set tpa_rate = 15
set tpa_breakpoint = 1560

# 
```

BavariaAirview, 2017-08-23:
Had this too. But can't reproduce. Was not Sure if I  had done something wrong.

EggsBenedict, 2017-08-23:
OH! creepy. Hmm one thing to point out is that I was unable to ever save things in the modes tab, so I did add some modes via CLI.  Perhaps I botched that?

> aux 0 0 0 1650 2100
> aux 2 1 3 900 1175
> aux 3 2 3 1300 1700

DanNixon, 2017-08-23:
This looks to be a duplicate of #3926

Can you retest with the most recent nightly.

EggsBenedict, 2017-08-23:
I can definitely try!
Is the latest nightly #111?
https://ci.betaflight.tech/job/Betaflight/111/artifact/obj/

EggsBenedict, 2017-08-24:
Is this really a duplicate? I don't have a prearm switch / failsafe switch set. I'm just using the same method I've used in the past (arm mode via swtich), which now appears to have a very different behavior. If using a prearm switch is now mandatory to avoid this situation that seems like a departure from previous versions. (I still haven't verified with the latest nightly). Perhaps I'm missing something though.

mikeller, 2017-08-24:
There is something weird going on with arming happening as soon as conditions are met, as opposed to requiring the arming mode being enabled **after** arming conditions are met. This issue and #3926 are both about this, and it will be beneficial for everybody to discuss on only one ticket.

EggsBenedict, 2017-08-24:
Oh ok. I get it now. retested with nightly listed above and the issue above appears to be resolved for me!
YIPPEEEEEEEEEEEEE!

Thanks very much for all your hard work and attentiveness. Sorry I didn't spot this as a duplicate.  Also really like the addition of the "prearm" mode, I'll definitely use that from now on.

sidenote: it's so weird ... this was just a fluke. I never run into issues like this but I was testing a failsafe so the arm switch was left on when I turned the TX back on.

closed 2017-08-24 by EggsBenedict