betaflight/betaflight #3956: F4 Flame 3.2 RC3 motor arm issue

rcspider, 2017-08-25:
Hi,

I've installed 3.2 RC3 version to F4 flame(Omnibus)

https://www.banggood.com/F4-FLAME-Race-Spec-STM32F405-Flight-Control-Built-in-BETAFLIGHT-OSD-5V-BEC-PDB-Current-Sensor-p-1136917.html?rmmds=search

Motors sometime starts, sometimes not. Sometimes start only 1 or 2. Version 3.1.7 works fine.
I think there is a problem.

BR

richard-scott, 2017-08-25:
Have you tried the latest 3.2 from here:
https://ci.betaflight.tech/job/Betaflight/lastSuccessfulBuild/artifact/obj/

jflyper, 2017-08-25:
## 3.2RC3 target OMNIBUSF4 has been successfully used by many users. You seem to have configuration problem, or mismatch with BFC.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://rcgroups.com/forums/showthread.php?t=2464844), [Slack](https://slack.betaflight.tech) or other user support forums & groups (e.g. facebook).

closed 2017-08-25 by jflyper

rcspider, 2017-08-25:
Hi,

Thanks for the answer (Richard). Latest firmware seems to be fine. No arming issues. I will try tomorrow on the site.

BR/Guven