betaflight/betaflight #3960: Drone at  full throttle flips (with blackbox log)

xracer666, 2017-08-25:
When i punch full throttle on my drone it sometimes flips for no reason. I checked blackbox log and it looka like motor 3 is stucking at 100%. I am new so i have no idea what to do or how to identify problem. Please help me fix this. 
This happens in 3rd and 14th throttle punch in log.
I have bf version 3.2.1 and my drone is stock wizard x 220 s

Link to blackbox log

https://wetransfer.com/downloads/b7cb62468af16eb2770193529e76afca20170825201804/4e283854ca9e995f7dc6d64d7b4c0aa720170825201804/c17374

Je47, 2017-08-25:
try with Airmode and Antigravity enabled, this may fix the problem

xracer666, 2017-08-25:
I have airmode on but not sure how to enable anti gravity

Je47, 2017-08-25:
modes tab

Je47, 2017-08-25:
also check your weight distribution/balance

xracer666, 2017-08-25:
Ill give it a try

etracer65, 2017-08-25:
Looks like a de-sync on motor 3.  In both cases motor 3 isn't producing enough (or any) thrust for a short period of time and the flight controller is commanding it to 100% to compensate.  Classic example of a de-sync.  If you're using oneshot or multishot try increasing your minimum throttle.  For DSHOT increase the idle percentage.

Also possible that motor 3 is beginning to have problems, the prop might be slipping (make sure it's tight), or that the ESC is having problems.

Lastly it could be ESC settings.  Make sure you're on the latest firmware and you might have to adjust the motor timing or demag compensation.  All this depends on your motor and ESC combination.

xracer666, 2017-08-26:
Iam using dshot600 and my idle throttle value is curently set to 4.5. can you sugest how much shuld i raise it?

xracer666, 2017-08-26:
Hopefully not one of those two. I crashed only once and i dd hit ground with motor 2 so i dont know what could couse motor or esc damage. Will also try anti gravity mode. I hope i can fix it in betaflight, without waitng 1 month for spare parts

brycedjohnson, 2017-08-27:
What escs are you using?

xracer666, 2017-08-27:
Stock blhelis 30 a 4in1

xracer666, 2017-08-27:
I am on vacation atm but as soon as i come back home ill try to switch motor 2 and 3 to see if problem will move to motor 2 . If it will still be on same place with difrent motor its esc problem i think. Hopefully i dont screw something up becouse i never did soldering before

xracer666, 2017-08-30:
Finally fixed it! Problem was in bheli configurator. Max throttle was set to 1830 by default. I raised it to 2000 and now its working great! Thx to everyone who helped me!

closed 2017-08-30 by xracer666