betaflight/betaflight #3962: LED strip conflicts with hexacopters on ANYFCF7

andrejpodzimek, 2017-08-26:
Only 5 out of 6 motors will work [on this board](https://www.aliexpress.com/item/New-Arrival-Realacc-F7-Pro-Betaflight-MPU6000-STM32F745VGT6-2-4S-BEC-5V1A-30-5X30-5mm-Flight/32812353160.html) when the LED strip is enabled, due to the DMA resource assignment. Without LED strip:

    DMA1 Stream 0: FREE
    DMA1 Stream 1: MOTOR 2
    DMA1 Stream 2: MOTOR 6
    DMA1 Stream 3: FREE
    DMA1 Stream 4: MOTOR 3
    DMA1 Stream 5: MOTOR 5
    DMA1 Stream 6: MOTOR 4
    DMA1 Stream 7: MOTOR 1
    DMA2 Stream 0: FREE
    DMA2 Stream 1: FREE
    DMA2 Stream 2: FREE
    DMA2 Stream 3: FREE
    DMA2 Stream 4: FREE
    DMA2 Stream 5: FREE
    DMA2 Stream 6: FREE
    DMA2 Stream 7: FREE

With LED strip:

    DMA1 Stream 0: FREE
    DMA1 Stream 1: MOTOR 2
    DMA1 Stream 2: MOTOR 6
    DMA1 Stream 3: FREE
    DMA1 Stream 4: MOTOR 3
    DMA1 Stream 5: LED_STRIP
    DMA1 Stream 6: MOTOR 4
    DMA1 Stream 7: MOTOR 1
    DMA2 Stream 0: FREE
    DMA2 Stream 1: FREE
    DMA2 Stream 2: FREE
    DMA2 Stream 3: FREE
    DMA2 Stream 4: FREE
    DMA2 Stream 5: FREE
    DMA2 Stream 6: FREE
    DMA2 Stream 7: FREE

Because this is an F7 that has 8 motor outputs, I was hoping that 6 of them could control motors and the two remaining ones could be used for a beeper (using [this hack](https://quadmeup.com/anyfc-f7-and-diy-buzzer-for-inav/)) and an LED strip. Is this possible or does it require some physical resoldering? I thought F7 wouldn't have as many timer / DMA limitations as F4s...

andrejpodzimek, 2017-08-26:
Forgot to add: I tried this also **without** dshot, on multishot. I thought it would remove the conflict. But nope, there are some odd motor twitches and shivers in multiple motors on multishot. With dshot + LED strip only one motor makes weird sounds, quite loud.

mikeller, 2017-08-26:
>  I thought F7 wouldn't have as many timer / DMA limitations as F4s...

Nope, the number of available timers / DMA channels is pretty much identical between F4 and F7.

Your setup should work without Dshot, do you have a scope to check what's going on on the channels when they twitch?

Looking at https://github.com/betaflight/betaflight/blob/dafdde5112e34fa64828e23ee5e9159346e0e869/src/main/target/ANYFCF7/target.c#L30-L40 , you should be able to put LED_STRIP onto one of the S3_IN to S6_IN input pins to make it work with a hex mixer and Dshot.

jflyper, 2017-08-26:
@andrejpodzimek You'll probably have to watch out UART TX DMA also. #3906

andrejpodzimek, 2017-08-26:
@jflyper You're right, it's not just the LED strip. It also conflicts with UART DMAs. Because as usual, I have too much stuff connected on the UARTS.

  1. free
  2. free
  3. OSD
  4. FrSky S.Bus
  5. FrSky SmartPort [in a hardware UART, unlike the F4 \o/ \o/ \o/]
  6. free
  7. GPS
  8. RunCam Split

    DMA1 Stream 0: SERIAL_TX 8
    DMA1 Stream 1: SERIAL_TX 7
    DMA1 Stream 2: FREE
    DMA1 Stream 3: SERIAL_TX 3
    DMA1 Stream 4: SERIAL_TX 4
    DMA1 Stream 5: FREE
    DMA1 Stream 6: FREE
    DMA1 Stream 7: SERIAL_TX 5
    DMA2 Stream 0: FREE
    DMA2 Stream 1: FREE
    DMA2 Stream 2: FREE
    DMA2 Stream 3: FREE
    DMA2 Stream 4: ADC
    DMA2 Stream 5: FREE
    DMA2 Stream 6: FREE
    DMA2 Stream 7: FREE

When I switch dshot on, it seems to conflict with UART3 (with one motor less in `resource list` and UART stuff listed instead) and **possibly** also with other UARTS.

I say possibly, because the board stops booting once I switch to dshot with the port configuration above. With an initial switch to dshot (right after re-flashing + full chip erase), dshot works, motors beep and work as expected, but then with a certain critical number of enabled UARTs it just makes the board unbootable. There's some LED flashing when I plug in the USB, but no USB communication.

Is actually multishot that much worse than dshot in terms of latency? Dshot is awesome as a concept, but after all this hassle I've had with it (e.g. [weird I2C trouble on my F4](https://github.com/betaflight/betaflight/issues/3861)), I'm started to think whether it's worth it. :disappointed:

andrejpodzimek, 2017-08-26:
@mikeller IIUC, the dshot + LED strip twitches are kind of understood, so you want me to look into the multishot + LED strip twitches. Hope that's correct.

The twitches occur on motor 5 (only). They are both audible and tangible, in the sense that the motor shivers at random (in a disarmed state). Most of the time the PWM channel looks like this:
[image]
From time to time there's something eventful that lasts half a millisecond, almost exactly:
[image]
Zooming into one of the 0.5ms signal bursts:
[image]

Resource mapping and DMAs, for the record:

    A00: MOTOR 6
    A01: MOTOR 3
    A02: MOTOR 2
    A03: MOTOR 4
    A04: MPU_CS
    A05: SPI_SCK 1
    A06: SPI_MISO 1
    A07: SPI_MOSI 1
    A08: FREE
    A09: FREE
    A10: FREE
    A11: USB
    A12: USB
    A13: FREE
    A14: FREE
    A15: FREE
    B00: FREE
    B01: FREE
    B02: BEEPER
    B03: FREE
    B04: FREE
    B05: LED_STRIP
    B06: LED 2
    B07: LED 1
    B08: MOTOR 1
    B09: FREE
    B10: I2C_SCL 2
    B11: I2C_SDA 2
    B12: FREE
    B13: FREE
    B14: FREE
    B15: FREE
    C00: ADC_BATT
    C01: ADC_CURR
    C02: FREE
    C03: FREE
    C04: MPU_EXTI
    C05: FREE
    C06: FREE
    C07: FREE
    C08: FREE
    C09: FREE
    C10: FREE
    C11: SERIAL_RX 4
    C12: SERIAL_TX 5
    C13: FREE
    C14: FREE
    C15: FREE
    D00: FREE
    D01: FREE
    D02: FREE
    D03: FREE
    D04: FREE
    D05: FREE
    D06: FREE
    D07: FREE
    D08: SERIAL_TX 3
    D09: SERIAL_RX 3
    D10: FREE
    D11: FREE
    D12: I2C_SCL 4
    D13: I2C_SDA 4
    D14: FREE
    D15: FREE
    E00: SERIAL_RX 8
    E01: SERIAL_TX 8
    E02: FREE
    E03: FREE
    E04: FREE
    E05: FREE
    E06: FREE
    E07: SERIAL_RX 7
    E08: SERIAL_TX 7
    E09: FREE
    E10: FREE
    E11: SPI_PREINIT
    E12: SPI_SCK 4
    E13: SPI_MISO 4
    E14: SPI_MOSI 4
    E15: FREE

    DMA1 Stream 0: SERIAL_TX 8
    DMA1 Stream 1: SERIAL_TX 7
    DMA1 Stream 2: FREE
    DMA1 Stream 3: SERIAL_TX 3
    DMA1 Stream 4: SERIAL_TX 4
    DMA1 Stream 5: LED_STRIP
    DMA1 Stream 6: FREE
    DMA1 Stream 7: SERIAL_TX 5
    DMA2 Stream 0: FREE
    DMA2 Stream 1: FREE
    DMA2 Stream 2: FREE
    DMA2 Stream 3: FREE
    DMA2 Stream 4: ADC
    DMA2 Stream 5: FREE
    DMA2 Stream 6: FREE
    DMA2 Stream 7: FREE

andrejpodzimek, 2017-08-26:
@mikeller @jflyper I forgot to mention that I'm still using [this hack](https://www.rcgroups.com/forums/showthread.php?2847666-Next-generation-flight-controller-AnyFC-F7/page5#post38147173) on the aircracft. That's because without the hack my baro is not detected and no CLI commands can fix it. :confused: With the hack both baro (I2C bus 4) and compass (I2C bus 2) work out of the box.

    System Uptime: 330 seconds
    Voltage: 156 * 0.1V (4S battery - OK)
    CPU Clock=216MHz, GYRO=MPU6000, ACC=MPU6000, BARO=MS5611, MAG=HMC5883
    SD card: Startup failed
    Stack size: 2048, Stack address: 0x20010000
    I2C Errors: 1, config size: 1975, max available config: 32768
    CPU:11%, cycle time: 129, GYRO rate: 7751, RX rate: 111, System rate: 9
    Arming disable flags: CLI

mikeller, 2017-08-27:
Looks like something is messing with the timers for MultiShot & LED_STRIP. :-( Since we are only starting out with F7 support, there are still a lot of hardware details that we have not yet completely figured out, and this seems to be one of them.

Re board lockups for UART DMA conflicts with other resources, I have seen this happen on other F7 boards, so this is another one from the 'we need to figure this out' pile. :-(

andrejpodzimek, 2017-08-27:
[This](https://github.com/betaflight/betaflight/issues/3906#issuecomment-325078429) is interesting. So I've just moved MSP/OSD from UART3 to UART1 and swapped S.Bus with SmartPort, so that SmartPort is on UART4, using only TX and S.Bus is on UART5, only using RX. I freed the UART4 RX and UART5 TX resources. I was hoping to get fewer conflicts or something like that. But the board no longer communicates over USB, so I'll have to reflash it. The motors do beep, but only the initial sequence, not the typical three `DSHOT600` beeps. Looks like the ESCs boot, but nothing talks to them. This repeats twice. (?!) And then it's stuck. There's no USB communication and it doesn't send telemetry. Going back to multishot for the time being.

andrejpodzimek, 2017-09-06:
So, I've just gone through some re-wiring hassle. Now I'm using only UARTs 1, 4, 6 and 8 for both TX and RX and UART 5 only for RX. But enabling dshot brings the board into an undefined state, just as before. :cry: The diff:

    # resources
    resource BEEPER 1 B09
    resource MOTOR 7 NONE
    resource MOTOR 8 NONE
    resource LED_STRIP 1 E06
    resource SERIAL_TX 2 NONE
    resource SERIAL_TX 3 NONE
    resource SERIAL_TX 5 NONE
    resource SERIAL_TX 7 NONE
    resource SERIAL_RX 2 NONE
    resource SERIAL_RX 3 NONE
    resource SERIAL_RX 4 NONE
    resource SERIAL_RX 7 NONE
    
    # mixer
    mixer HEX6X
    
    # feature
    feature GPS
    feature TELEMETRY
    feature ANTI_GRAVITY
    
    # serial
    serial 0 1 115200 57600 0 115200
    serial 3 32 115200 57600 0 115200
    serial 4 64 115200 57600 0 115200
    serial 5 2 115200 57600 0 115200
    serial 7 16384 115200 57600 0 115200
    
    # aux
    aux 0 0 0 1700 2100
    aux 1 1 1 900 1625
    aux 2 2 1 1625 1875
    aux 3 3 1 900 1375
    aux 4 5 6 1700 2100
    aux 5 11 1 900 1125
    aux 6 13 3 1300 2100
    aux 7 28 4 1700 2100
    
    # adjrange
    adjrange 0 0 2 900 2100 12 2
    
    # rxfail
    rxfail 5 s 1000
    rxfail 6 s 1000
    rxfail 7 s 2000
    
    # master
    set align_mag = CW270FLIP
    set mag_hardware = HMC5883
    set mag_declination = 2180
    set magzero_x = -124
    set magzero_y = -54
    set magzero_z = 411
    set baro_hardware = MS5611
    set fpv_mix_degrees = 20
    set blackbox_device = NONE
    set min_throttle = 1030
    set dshot_idle_value = 200
    set motor_pwm_protocol = MULTISHOT
    set current_meter = ADC
    set battery_meter = ADC
    set ibata_scale = 366
    set gps_provider = UBLOX
    set gps_sbas_mode = EGNOS
    set gps_auto_baud = ON
    set pid_process_denom = 1
    set frsky_unit = METRIC

But at least I have a fancy new beeper, because I've managed to [install the transistor](https://quadmeup.com/anyfc-f7-and-diy-buzzer-for-inav/) :relieved:. I have it on motor output 7 now. It was on B02 by default and I have no idea where that is on this board.

In the diff above, I tried to disable all I'm not using, but no way, dshot just won't work in this configuration.

andrejpodzimek, 2018-01-08:
As already mentioned under #3861, the fancy new DMAs and timers for DShot may make it feasible to have not just DShot and I2C, but also LEDs on the same build. At least DShot + I2C work for me without issues now on current master. My `ANYFCF7` hexacopter may get its nocturnal flight rating at last!

mikeller, 2018-01-08:
@andrejpodzimek: Can this be closed then?

closed 2018-04-15 by andrejpodzimek