betaflight/betaflight #3963: Runcam swift 2 BF OSD not displaying only when in 3.2

kameo4242, 2017-08-26:
I use a runcam swift 2 on a AIO board HGLRC F4 V5PRO (omnibus).

With BF 3.1.7, I can control the OSD from BF and the one from the camera doesn't prempt the OSD.

With BF 3.2 RC, the Runcam OSD is the only thing I get and can't view the one from BF.
(I tried switch PAL / NTSC in BF)

Downgrading to BF 3.1.7 solved the problem.

jflyper, 2017-08-26:
@kameo4242 

OSD is the overlay. CMS is the menu system invoked by a stick command.

Can describe what is happening using these words?

kameo4242, 2017-08-27:
Sorry, 1000 times sorry. Got grounding issues I think. Will review all the grounds and contacts, reflash 3.2 and check again. Looks like a false positive, my bad.

spuder, 2017-09-08:
Is this issue resolved? Would you mind closing the github issue @kameo4242

kameo4242, 2017-09-09:
yep it is. I can confirm this was rather a wiring / grounding problem, that just has been tirggered at the same time as the update. Sorry for the false positive.

Great work on BF3.2 guyz, many thanks

closed 2017-09-09 by kameo4242