betaflight/betaflight #3967: problem with spracing f4 evo and hm10 bluetooth since BF3.2rc3

matoy, 2017-08-26:
Hi guys

I have an sp racing f4 evo FC and a HM10 bluetooth module.
i wire HM10 RX/TX to UART1 and set the port to MSP and correct baud in configurator.

In BF 3.1.7, i can connect with bluetooth on a mobile device (android or iphone) to the FC. Everything works well.
With new version BF3.2rc3, exact same setup doesn't work anymore. (tried to default the conf and reconfigure port but that didn't help).

NB1: i tried with any version of cleanflight from 2.0 to 2.1 and could not make it work.
NB2: at first i wanted to use UART3 but it seems PPM doesn't work anymore, don't know why. 

Thank you and don't hesite to ask more details if required.

mikeller, 2017-08-27:
Hi @matoy. You wouldn't have a logic analyser to be able to check if any data is transferred between your bluetooth module and the flight controller in 3.2, and if so, what baud rate it is at?

matoy, 2017-08-27:
Hi,

I don't advanced skills like many people here so I don't really understand your request. :-)

I just wanted to inform you about this not working anymore since BF3.2. And OK with 3.1.7. But I couldn't make it work on any version of cleanflight...

Since I have another issue with this FC: I can't use PPM and anything else on uart3 (gps, bluetooth module or osd) at the same time. It's not normal because PPM uses uart2, not 3.

A guess is that my FC (a chinese clone) might have a problem...

If somebody can confirm that he hasn't the issue, we can can close the issue.

jflyper, 2017-08-27:
@matoy Can we see outputs from CLI command "diff"?

matoy, 2017-08-27:
Hi,

You can find these to diff output: the one working with 3.1.7 and the other one not working with 3.2.0rc3

[diff not working.txt](https://github.com/betaflight/betaflight/files/1254641/diff.not.working.txt)
[diff working.txt](https://github.com/betaflight/betaflight/files/1254642/diff.working.txt)

the HM10 module is on UART4:
serial 3 1 19200 57600 0 115200
Is it normal that UART4 states serial3 ? i got this on both versions.

I bought this clone:
https://www.aliexpress.com/item/F21746-JMT-SP-Racing-F4-EVO-Flight-Controll-2-6S-Built-in-BEC-Barometer-Voltage-Sensor/32824536522.html
By the way: the S/R bridge was not soldered by defaults like it should have been (that's in the EVO manual on sp racing web site)... It took me time to make receiver work ;-)

thank you.

jflyper, 2017-08-27:
Okay... I flashed my Nucleo-F405RG with SPRACINGF4EVO and loaded your diff.not.working. I see bunch of garbage coming out from UART4. What is this ...

jflyper, 2017-08-28:
@matoy 
Disabling OSD in config tab makes UART4 available for MSP. Can you try this?

jflyper, 2017-08-28:
@hydra This one is for you. It's caused by proactive MSP messages generated when OSD feature is enabled on F4EVO, either flooding the 19200 bps BT link or causing MSP client that can not handle incoming MSP messages that do not correspond to an outstanding request to fail.

matoy, 2017-08-28:
hi,

i confirm that disabling OSD feature solves the problem.
i don't know if you consider there's an issue there or if it's "normal"...

@jflyper : thank you.

hydra, 2017-08-31:
I'm not supporting anyone that buys clone boards.  If someone else reports the problem I can look into it.

closed 2017-08-31 by hydra

mikeller, 2017-08-31:
@matoy: It is the expected behaviour, which is a shortcoming of how the OSD feature was implemented. Do you actually need the external OSD? If so, you should consider buying an original SPRacing board, since they are the only ones supporting this feature.

hydra, 2017-08-31:
supported: genuine F3EVO + F3OSD without bluetooth, F4EVO + F3OSD without bluetooth.  
unsupported: any clone board, F3EVO or F4EVO + bluetooth with OSD enabled when not OSD is connected - solution: disable OSD feature.
if you want to use any board + F3OSD + bluetooth then please create a feature request and reference this issue.

hydra, 2017-08-31:
also, to further clarify, it's a shortcoming of not how the OSD feature was implemented, but how the CMS MSP displayPort code was implemented.  The OSD feature just leveraged that existing MSP displayPort code.

specifically the issue is with the old `mspSerialPush` function:

https://github.com/cleanflight/cleanflight/blob/master/src/main/msp/msp_serial.c#L276-L301

hydra, 2017-08-31:
see https://github.com/cleanflight/cleanflight/pull/2934