betaflight/betaflight #3970: DYS F4 beeper stopped working after 3.2 upgrade

bluehallu, 2017-08-27:
Beeper was working fine under 3.1.7. 

After upgrading to 3.2, beeper doesn't work for either beeper mode on switch nor for all the normal operations (power up, accelerometer calibrated etc). The only sound I ever hear now are the ESCs.

brycedjohnson, 2017-08-27:
Did you just do a copy and paste of the diff?  You may need to reset to default configure through the configurator.  

Also post your CLI diff so people may better help you.

jflyper, 2017-08-27:
@bluehallu Are you using DYSF4PRO target?

bluehallu, 2017-08-27:
I am using the DYSF4PRO target. I did not copy paste the diff. I saved the output of diff all in a text file then applied every change via Configurator (except small_angle).

AndersHoglund, 2017-08-27:
My DYSF4PRO beeps just fine, using a fairly old BF3.2.0 build.
`# Betaflight / DYSF4PRO (DYS4) 3.2.0 Aug 14 2017 / 13:24:51 (8995dce)`

Updated to the current HEAD, still beeps OK.
`# Betaflight / DYSF4PRO (DYS4) 3.2.0 Aug 27 2017 / 10:37:56 (d4d3711)
`

jflyper, 2017-08-27:
I also tested the DYSF4PRO with RC3 and current master, and the beeper works just fine.

@bluehallu Does the beeper ever beeps or make some noise upon power up? If it doesn't, I think you have a hardware problem.

And, I've noticed that your beeper mode entry is very weird. Did you do this intentionally?
```
aux 2 13 1 1975 2025
```

For further assistance, please redirect your problem statement to other places on the net.

Ref:
https://www.rcgroups.com/forums/showthread.php?2856541-DYS-F4-OSD-BEC-AiO-Flight-controller

closed 2017-08-27 by jflyper

bluehallu, 2017-10-01:
Beeper was dead indeed, installed a new one and it works. I guess it died on the flight before going home and flashing 😅