betaflight/betaflight #3976: Beeper and status led blinking

Antoineplayer84, 2017-08-27:
Hello, 

One week ago, I soldered my rx to uart 3 because it takes less space and I had no problem. I hovered to test if everything was good and I had no issues. 

But today, I wanted to fly again (I didn't fly since I soldered my rx) and my quad fly, it respond correctly; but the status led is blinking quickly and my buzzer is beeping quickly too. 

I didn't find why it's doing that because it fly well. I could turn off the beeper but if it's beeping there is a reason and I would find it. 

One time I plugged the Lipo, it was beeping like a code I don't remember exactly but i think it was like 5-6 long beep then 2 very short beep.

Video : https://youtu.be/CTrKuqFg3xo 

Thank you for your help

AndersHoglund, 2017-08-27:
Low battery
https://github.com/cleanflight/cleanflight/blob/master/docs/Buzzer.md#tone-sequences

Antoineplayer84, 2017-08-31:
Oh thank you  I didn't calibrate the Vbat

closed 2017-09-01 by mikeller