betaflight/betaflight #3980: FC Reboot

Lendvi, 2017-08-27:
In RC3 when i turn on beeping(via dshot esc), them arm, when the fc is armed erase blackbox log via mods switch amd then disarm, fc stops working and after few second reboots. Board SPRF3

mikeller, 2017-08-28:
Can you please supply the flight controller model and the output of `diff`?

Lendvi, 2017-08-28:
SPRF3 bought from rotorama.cz 

CLI

# diff
# Betaflight / SPRACINGF3 (SRF3) 3.2.0 Aug  6 2017 / 13:37:51 (981df1d54)

feature -RX_PPM
feature -RSSI_ADC
feature -TRANSPONDER
feature RX_SERIAL
feature SOFTSERIAL
map TAER1234
serial 1 32 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200
aux 0 0 0 900 1200
aux 1 13 3 1800 2100
aux 2 26 2 900 1200
aux 3 28 1 1800 2100
aux 4 31 2 1800 2100
aux 5 35 1 900 1200
set gyro_sync_denom = 2
set acc_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = SBUS
set dshot_idle_value = 300
set motor_pwm_protocol = DSHOT600
set current_meter = NONE
set pid_process_denom = 1
profile 0

set dterm_lowpass_type = PT1
set dterm_lowpass = 90
rateprofile 0

set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 62

#

DooDoo512, 2017-09-04:
Low your loop time and try...
Such as 1K gyro/1K PID , even lower