betaflight/betaflight #3981: Airmode not working correctly?

rossonado, 2017-08-27:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://rcgroups.com/forums/showthread.php?t=2464844), [Slack](https://slack.betaflight.tech) or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1.) Describe your problem;
2.) Include ways to reproduce the problem;
3.) Provide information about your flightcontroller and other components including how they are connected/wired;
4.) Add the used configuration and firmware version;
5.) Create a `diff` and post it here in a code block. Put \`\`\` (three backticks) at the start and end of the `diff` block; and
6.) Remove this Text :).

rossonado, 2017-08-27:
3 different quads flashed to 3.2rc3, heavy freestyle rig, with matek f405 rcx 2206v3, Spedix is25, superlight 4 inch, dys f4, rotorgeeks 1407, Spedix 20amp, superlight 3 inch hglrc xjb 415 f4.

All setup for Dshot 600 with 8k/8k

After punchouts on all three quads, at zero throttle the quads drift, then nose down as though doing a split s on their own, happens while inverted too. I've played with I, p, anti gravity, digital idle speed and nothing I can do seems to have much effect. I have removed filters, set to PT1 and am using dynamic filters (filters were also removed on 3.17), but there is zero vibration and motors are cold. Fly fine under throttle. Am I missing something here? All three fly fine on 3.17.

Thx

RipperDrone, 2017-08-28:
See #3723

bkeinert, 2017-08-28:
Can you try RC4 and maybe try increasing iterm_limit? Not exactly sure, but that could probably help.

ee0u30eb, 2018-04-10:
Has this issue been resolved? I am experiencing identical issues on Betaflight 3.3 with a Matek F405 FC. As per above I have tried all manner of PIDs and massive Anti Gravity but to no avail. I have posted elsewhere and uploaded the following blackbox and flight video: https://youtu.be/XyO19MVlrRk 
Blackbox log which accompanies that video is attached.
[LOG00008.zip](https://github.com/betaflight/betaflight/files/1895999/LOG00008.zip)

closed 2018-05-17 by stale[bot]