betaflight/betaflight #3982: LED strip on N-Channel mysteriously turns on non-N-Channel output

jflyper, 2017-08-28:
- When a pin designated as complementary output (N-Channel) of a timer channel is configured as LED strip output, a corresponding non-complementary (normal) output mysteriously acquire a pin for output. 

- This has been observed with F4 with PB0 as TIM1_CH2N and PA9.

PA9 (TIM1_CH2) is configured for LED_STRIP. No effect on PB0
[image]

PB0 (TIM1_CH2N) is configured for LED_STRIP, with `TIMER_OUTPUT_INVERTED`. PA9 is mysteriously configured as output, with the same output as PB0.
[image]

PB0 (TIM1_CH2N) is configured for LED_STRIP, with `TIMER_OUTPUT_N_CHANNEL`.  PA9 is mysteriously configured as output, and is showing the correct wave form for the LED.
[image]

PB0 (TIM1_CH2N) is configured for LED_STRIP, with `TIMER_OUTPUT_N_CHANNEL | TIMER_OUTPUT_INVERTED`. PA9 is mysteriously configured as output, and is inverted.

[image]

- This doesn't happen on motor outputs.

[image]

- And LED_STRIP on complementary channel does not work (this is a separate issue).

jflyper, 2017-08-28:
Interesting.

There is a sequence in `drivers/light_ws2811strip_stdperiph.c`:
https://github.com/betaflight/betaflight/blob/v3.2.0-RC4/src/main/drivers/light_ws2811strip_stdperiph.c#L118-L119
```
118     TIM_CCxCmd(timer, timerHardware->channel, TIM_CCx_Enable);
119     TIM_Cmd(timer, ENABLE);
```

Some peeking into GPIOA's AFR revealed that AF1 (TIM1_CH2) is turned on for PA9 during the call to `TIM_Cmd` on line 119.
Since `TIM_Cmd` is a simple bit op on CCER:
https://github.com/betaflight/betaflight/blob/v3.2.0-RC4/lib/main/STM32F4/Drivers/STM32F4xx_StdPeriph_Driver/src/stm32f4xx_tim.c#L592-L608
```
    TIMx->CR1 |= TIM_CR1_CEN;
```
it is assumed that, if a timer does not have a valid output pin, then a pin is automagically selected and configured with appropriate AF.

jflyper, 2017-08-28:
As a side note, if the line 118 above is modified so that `TIM_CCxNCmd` is called for N-Channel, then the automagic AF assignment on PA9 does not occur, and the LED strip itself starts to work.

ledvinap, 2017-08-28:
@jflyper : So value of AF register is actually changed when timer is enabled?

I can imagine that output multiplexer is actually implemented in 'forward' way, selecting PIN for for module output (and not module connected to PIN). This may have consequences when collision occurs (connecting two pins to single peripheral) ...

Using TIM_CCxCmd on N channels looks like bug in code ...

jflyper, 2017-08-28:
@ledvinap 
> So value of AF register is actually changed when timer is enabled?

I can't reproduce it under the debugger. Could have been my bad code that tried to copy AFR into `debug` variable.

> Using TIM_CCxCmd on N channels looks like bug in code ...

Yes.

As `timerInit` initializes all entries in `timerHardware` array to timer AF, calling `TIM_CCxCmd` explain the activation of P channel.

jflyper, 2017-08-29:
Closed via #3993

closed 2017-08-29 by jflyper