betaflight/betaflight #3985: CLI DFU

Gaz-B, 2017-08-28:
Could we have the CLI DFU command working again.

mikeller, 2017-08-28:
It is working, it is called `bl` (for BootLoader) now.

Gaz-B, 2017-08-28:
But at the top of the config it's still called dfu. I say if it's not broken don't fix it. Developers

mikeller, 2017-08-28:
Where does it say dfu?

```
Entering CLI Mode, type 'exit' to return, or 'help'

# help
[...]
bl - reboot into bootloader
[...]
```

Looks all ok to me.

lichtl, 2017-08-28:
I suspect he is referring to the "DFU" interface in the configurator

mikeller, 2017-08-28:
Well, the DFU interface is shown in the configurator for boards that use VCP and have a DFU mode. The rename was triggered by users with boards with a CP210x that `dfu` did not put their board into DFU mode...

Gaz-B, 2017-08-28:
At the top of the configurator where it normally shows com port. When I had removed enough to get to the boot button.

From: Michael Keller
Sent: Monday, 28 August, 10:51
Subject: Re: [betaflight/betaflight] CLI DFU (#3985)
To: betaflight/betaflight
Cc: Gaz-B, Author

Well, the DFU interface is shown in the configurator for boards that use VCP and have a DFU mode. The rename was triggered by users with boards with a CP210x that dfu did not put their board into DFU mode...

mikeller, 2017-08-28:
@Gaz-B: See above, 'DFU' is only shown there for targets that use DFU mode (i.e. they must have VCP). For other targets, the COM port is shown when they are in boot loader mode.

closed 2017-08-28 by mikeller

Gaz-B, 2017-08-28:
bl works on my Kiwi v2 but not the V1 just flashed the very latest build onto it and still no. It does something as configurator goes back to main page and after about 30secs I have a USB unidentifiable pop up. Power cycling the fc gets it all back to normal.

mikeller, 2017-08-29:
Ok, that makes me suspect that it's a driver problem. The only difference between KIWIF4 V1 and V2 in terms of USB connection is the USB product string that is returned by the flight controller. This will cause the host OS to associate drivers with the two different versions of the board individually, and the association with V1 on your computer seems to be broken. Do you have a different computer that you could try on, to verify this theory?

Gaz-B, 2017-08-29:
Yes I will try it.

Gaz-B, 2017-08-31:
The behaviour is exactly the same on both PCs the V2 goes into DFU from the BL command and the V1 doesn't.

AndersHoglund, 2017-08-31:
You have read the "###Troubles Entering Bootloader Mode (DFU)" -section here?: https://github.com/betaflight/betaflight/wiki/KIWIF4#faq--known-issues

It is not unique for this FC to behave in a mysterious way if there are signal sources connected to a UART etc. The bootloader catches not only USB-DFU but also UART and I2C signaling. I had for example problems with Spektrum SPM4648 sat connected and in auto bind mode, it disturbed bootloader USB-DFU detection. Had to disconnect the sat temporarily to be able to flash FC.

Gaz-B, 2017-08-31:
I have now tried disabling all uarts and no difference. Just to clarify it's fine if i push the boot button.

Gaz-B, 2017-08-31:
> "disabling all uarts" ?? As in disconnecting all wires or by deselecting UARTs in BFC??
>
> OK. If it is fine when entering BootLoader via the boot button, then this might be a case of BF firmware leaving the IO-pins in a state that is NOT the same as with a HW-reset, when exiting back to bl. And then the bootloader hooks on to this state. No idea why this would be different with a V1 or V2 FCs, randomness or some hardware differencies.

Deselecting no  battery so smart audio not connected only the sat rx connected and not autobind.

AndersHoglund, 2017-08-31:
"...only the sat rx connected.." Have you tried disconnecting?

Gaz-B, 2017-08-31:
I can confirm if either FC has a satellite receiver BL doesn't function.

jflyper, 2017-09-01:
It's a normal behavior.

@AndersHoglund 
I think we should put a note on the flasher page that chatty devices on UARTs may prevent the MCU to go into the boot loader mode in the expected way.

mikeller, 2017-09-01:
Good idea, @jflyper. Are **all** UARTS tested by the bootloader / DFU, or only certain ones?

AndersHoglund, 2017-09-01:
Funny thing is that there seems to be no problem if entering boot loader via boot button, only when returning from the firmware. Or might that be caused by different test conditions, like Tx on or off generating data traffic etc etc. Have not done any full permutations tests on this....

Usually UART1 and 2 are checked for traffic, depends on chip type. Check table 3 here: https://www.google.se/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=0ahUKEwjGsM2S4YPWAhXFNpoKHeygAu0QFggoMAA&url=http%3A%2F%2Fwww.st.com%2Fresource%2Fen%2Fapplication_note%2Fcd00167594.pdf&usg=AFQjCNFeZkqkh-OTIfyRSGM8__FBCfs-CA

AndersHoglund, 2017-09-01:
Could active devices on I2C and SPI also cause similar problems? Like SPI-Rx etc....

Delicate problem, a HW design issue really. All attached devices should be powered from a ON/OFF switch-able power source. Being OFF by default and in reset state,  powered on in a controlled fashion during firmware init.

jflyper, 2017-09-02:
> Could active devices on I2C and SPI also cause similar problems? Like SPI-Rx etc....

Under certain conditions, I think they would. I might have a look into how these work. At least there is no chatty device on I2C...

> Funny thing is that there seems to be no problem if entering boot loader via boot button

This can be explained by the fact that some receivers do not become chatty until they are bound to TX for the first time after power cycling.

jflyper, 2017-09-02:
I2C is unlikely to interfere. MCU will be slave and there must be a master that talks the appropriate protocol.

Some references:

[AN2606 STM32 microcontroller system memory boot mode](http://www.st.com/content/ccc/resource/technical/document/application_note/b9/9b/16/3a/12/1e/40/0c/CD00167594.pdf/files/CD00167594.pdf/jcr:content/translations/en.CD00167594.pdf)

[AN4221 I2C protocol used in the STM32 bootloader](http://www.st.com/content/ccc/resource/technical/document/application_note/4c/68/fe/72/a8/cd/47/83/DM00072315.pdf/files/DM00072315.pdf/jcr:content/translations/en.DM00072315.pdf)

AndersHoglund, 2017-09-02:
SPM4648 in Autobind mode seems to be chatty at power on, and no Tx of course.

Gaz-B, 2017-09-02:
My satellite isn't auto bind and no tx.

AndersHoglund, 2017-09-02:
I can not reproduce this anymore, Tx on or off. Autobind is not chatty, checked with scope.
BFF3 with 3.2.0-RC4, SPM4648 on UART2 Rx, SPEKTRUM2048.
Must be a combination of some other things.