betaflight/betaflight #3988: Add sticks overlay to in flight OSD

JonnyPhenomenon, 2017-08-28:
reference to #https://github.com/betaflight/betaflight-configurator/issues/630

Feature request:  **For the betaflight OSD**

Add the sticks to the OSD, kind of like the blackbox explorer can do, but in the FPV feed.

Sure, the pilot really would have no need for this, as they would be watching where they were going, but this feature would be good for "passengers" monitoring your flight, especially in training scenarios, where an instructor would be able to watch the control inputs and coach the pilot, or while demonstrating a flight while showing the movements of the sticks...

For example: (apologies for the crudeness of my sketch)
[image]

Note: it looks like someone has already done this with the tinyOSD - can it be replicated with the BF OSD without stepping on anyones toes? (Fishpepper)
http://fishpepper.de/2017/05/24/tinyosd-0-4g-full-betaflight-integration-early-preview-demo/

DanNixon, 2017-09-08:
It would be nice to have in BFOSD too, it won't have the same resolution as tinyOSD though. I'll have a look.

pkruger, 2018-12-03:
PR: https://github.com/betaflight/betaflight/pull/7167

Docteh, 2019-04-22:
@mikeller can this be closed? Stick overlay is in released version of betaflight now

JonnyPhenomenon, 2019-04-22:
I am ok with that.  Although, is the feature actually ready to go? or is there custom fonts that I would need to add to use it right now?

Docteh, 2019-04-22:
You have to flash a font from the latest configurator. 10.5.1

closed 2022-05-08 by haslinghuis