betaflight/betaflight #3989: BF 3.2 RC3 OSD Cell Voltage - Kakute F4 AIO

jcox303, 2017-08-28:
I'm having an issue with the reported cell voltage in the OSD. Min/Max cell voltages are set correctly in the GUI, but the OSD is showing 3.5v for a fully charged 4 or 5S battery. In the original BF 3.2 RC1 the value was correct. The main (total) batt voltage is correct and the mah draw has been calibrated. FC is Kakute F4 AIO.

Also, it's not only the OSD but Telemetry too as my Taranis telemetry shows the same data. I searched through the CLI commands and found report_cell_voltage = OFF as the default, changed to on and it was correct until the FC reboots again.

Bug or am I missing something?

\\\# Betaflight / KAKUTEF4 (KTV1) 3.2.0 Aug 26 2017 / 16:36:39 (779be8d)

# feature
feature RSSI_ADC
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# serial
serial 0 32 115200 57600 0 115200

# aux
aux 0 0 0 1800 2100
aux 1 26 0 1800 2100

# master
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set baro_hardware = NONE
set rc_interp = MANUAL
set rc_interp_int = 16
set blackbox_device = SPIFLASH
set motor_pwm_protocol = DSHOT600
set vbat_max_cell_voltage = 42
set current_meter = ADC
set ibata_scale = 460
set pid_process_denom = 1
set tlm_inverted = ON
set tlm_halfduplex = OFF
set report_cell_voltage = ON
set osd_units = IMPERIAL
set osd_rssi_alarm = 85
set osd_tim1 = 2561
set osd_tim2 = 2560
set osd_vbat_pos = 2416
set osd_rssi_pos = 2113
set osd_tim_1_pos = 2135
set osd_tim_2_pos = 33
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_crosshairs = 200
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 2400
set osd_mah_drawn_pos = 2424
set osd_craft_name_pos = 0
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 0
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_avg_cell_voltage_pos = 2407
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 0
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_bbox = OFF
set osd_stat_endbatt = ON
set osd_stat_bb_no = OFF
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set vcd_video_system = 2

# profile
profile 0

set dterm_notch_hz = 0
set p_pitch = 65
set i_pitch = 60
set d_pitch = 18
set p_roll = 43
set d_roll = 10
set p_yaw = 77

# rateprofile
rateprofile 0

set rc_rate = 85
set rc_rate_yaw = 85
set roll_srate = 83
set pitch_srate = 83
set yaw_srate = 83
set tpa_rate = 15
set tpa_breakpoint = 1750\\\

mikeller, 2017-08-29:
I tested with a KAKUTEF4 board, and the voltage readout after flashing 3.2.0-RC4 and resetting to defaults is correct. Try resetting to defaults, and if that does not help, your board is probably defective.

jcox303, 2017-08-30:
Upgraded both Kakute's to RC4 - confirmed issue is fixed. Thanks!

closed 2017-08-30 by jcox303