betaflight/betaflight #3990: Quad will ARM when plug in battery with ARM switch active during power up.

bnn1044, 2017-08-28:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://rcgroups.com/forums/showthread.php?t=2464844), [Slack](https://slack.betaflight.tech) or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1.) Describe your problem;
2.) Include ways to reproduce the problem;
3.) Provide information about your flightcontroller and other components including how they are connected/wired;
4.) Add the used configuration and firmware version;
5.) Create a `diff` and post it here in a code block. Put \`\`\` (three backticks) at the start and end of the `diff` block; and
6.) Remove this Text :).

DanNixon, 2017-08-28:
Duplicate of #3937.

Try with RC4 and report back if you still have an issue.

closed 2017-08-28 by mikeller

claytonbakerjr, 2017-09-14:
Hi I know this is closed but I am trying to zero in on the fix.
A friend has a CL Racing F4 loaded with BF3.2 RC4 using
a Spektrum DX6 and the same happens to him.
I've read through #3937 #3926 and from what I gather
this should have been resolved in RC4? 
Sorry to drag this out again

DanNixon, 2017-09-14:
@claytonbakerjr  What RX and RX protocol.

I believe @AndersHoglund tested the new arming checks with Spektrum gear and it worked fine for him.

claytonbakerjr, 2017-09-14:
Thanks @DanNixon unfortunately this guy is at MGP Nationals right now LOL so getting
exact specifics is difficult. I told him to be mindful of his arm switch until we can actually
dig deeper into his situation. If after we do put this through the testing and find that
it seems to have presented itself again I will have him post it up.
Appreciate the reply

AndersHoglund, 2017-09-14:
Did I test this? Geting old and senile :-) 
Anyhow, 3.2.0-RC5 works fine with only simple single arming on AUX switch. No pre-arm. Using Dx18G1, DYSF4PRO  and SPM4649T telemetry Rx on UART Tx3, SRXL .
If I powerup the rig with Tx AUX switch in ARMED  state, I get warning beeps and have to toggle the arm switch off and on again to arm. As it is supposed to be. 

Please, no more reports on old RCs. If you think there is a issue, retest on latest RC and report based on that. If possible, a re-test on the latest CI build would also be very good.

DanNixon, 2017-09-15:
@AndersHoglund I thought you did, maybe I'm mistaken. I have a feeling it was back when the additional checks for the initial receiver state were added (RX failsafe, bad initial ARM switch position, etc.).