betaflight/betaflight #3994: SRF3 - can't arm

ABLomas, 2017-08-28:
A bit strange issue and i do not have another board to verify.
< 3.0 BF - OK, no problems
3.2 RC1 - fine with 2/2K, not tested with 1/1K
3.2 RC3 - does not work with 1/1K, works OK with 2/2K
3.2 RC4 - does not work with 1/1, 2.2K

So, in short - Spracing F3 with SRF3 firmware does not work (can't arm) when is set to 1K gyro/1K PID loop. With 2/2 and RC3 it works OK. With RC4 it does not work at all, most of time.

With RC3 and gyro/pid set to 1/1K i see this in "status" output:
Arming disable flags: 0x224
With RC4 and gyro/pid set to any values i see this:
Arming disable flags: 0x1004

When i try to arm - it beeps 6 times (6 long with 6 short) which does not make sense to me (throttle too high? But it's def below min_check and above min_command).

In "sensor" tab everything looks OK - gyro and acc responds to movement, values does not look out of range. The same configuration (with minor adjustments between versions) works fine on earlier versions.

My current config diff:
\\\
name PR6
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 A03
resource SERVO 2 A02
servo 0 1000 2000 1500 100 -90
servo 1 1000 2000 1500 100 -90
servo 2 1000 2000 1500 100 -90
servo 3 1000 2000 1500 100 -90
servo 4 1000 2000 1500 100 -90
servo 5 1000 2000 1500 100 -90
servo 6 1000 2000 1500 100 -90
servo 7 1000 2000 1500 100 -90
smix reverse 0 1 r
smix reverse 0 3 r
smix reverse 0 4 r
smix reverse 0 6 r
smix reverse 0 8 r
smix reverse 0 10 r
smix reverse 1 1 r
smix reverse 1 3 r
smix reverse 3 6 r
smix reverse 3 8 r
smix reverse 3 10 r
smix reverse 4 1 r
smix reverse 4 3 r
smix reverse 4 4 r
smix reverse 4 6 r
smix reverse 4 8 r
smix reverse 4 10 r
smix reverse 7 3 r
smix reverse 7 4 r
smix reverse 7 6 r
smix reverse 7 8 r
smix reverse 7 10 r

feature -RX_PPM
feature -TELEMETRY
feature -RSSI_ADC
feature -TRANSPONDER
feature RX_SERIAL
feature MOTOR_STOP
feature CHANNEL_FORWARDING
feature DYNAMIC_FILTER
map TAER1234
serial 1 64 115200 57600 0 115200
serial 2 1 115200 57600 0 115200
aux 0 0 2 1300 2100
aux 1 2 4 1700 2100
aux 2 4 2 1700 2100
aux 3 13 4 1700 2100
aux 4 26 4 1300 1700
aux 5 28 2 1700 2100
aux 6 30 2 1700 2100
set gyro_sync_denom = 2
set gyro_lowpass_hz = 80
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 1
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 1
set baro_hardware = NONE
set rssi_channel = 8
set fpv_mix_degrees = 10
set serialrx_provider = SBUS
set blackbox_p_denom = 8
set vbat_scale = 114
set ibata_scale = 525
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 4
set debug_mode = NOTCH
profile 0

set dterm_lowpass_type = PT1
set dterm_notch_hz = 140
set dterm_notch_cutoff = 120
set setpoint_relax_ratio = 70
set dterm_setpoint_weight = 45
set yaw_lowpass = 80
set p_pitch = 48
set i_pitch = 52
set d_pitch = 16
set p_roll = 45
set i_roll = 41
set d_roll = 17
set p_yaw = 50
rateprofile 0

set rc_rate = 125
set rc_rate_yaw = 120
set rc_expo = 15
set rc_expo_yaw = 15
\\\

ABLomas, 2017-08-28:
Some additional information: friend verified this on another board, at least string "Arming disable flags: 0x1004" stay here. String changes with frequency - 1004 with 4/4, 4/2, 2/2, but with 1/1 we have "Arming disable flags: 0x1044".
Guess this is not my boards fault...

ABLomas, 2017-08-28:
For easier testing - just flash RC4 and this minimal diff shows issue (SBUS RX + beeper required):
```
# Betaflight / SPRACINGF3 (SRF3) 3.2.0 Aug 28 2017 / 12:08:46 (b2cd7294e)

feature -RX_PPM
feature -TRANSPONDER
feature RX_SERIAL
map TAER1234
serial 1 64 115200 57600 0 115200
aux 0 0 2 1300 2100
set serialrx_provider = SBUS
profile 0

rateprofile 0
```
(everything else is on defaults)
Also, with TX/RX turned on:
```
# status
System Uptime: 69 seconds
Voltage: 118 * 0.1V (3S battery - OK)
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n
Stack size: 2048, Stack address: 0x10002000
I2C Errors: 3, config size: 2001, max available config: 4096
CPU:8%, cycle time: 507, GYRO rate: 1972, RX rate: 99, System rate: 9
Arming disable flags: 0x1020
```

swilkens, 2017-08-28:
Somewhat related question; how do we process that arming disable flag to user understandable things? it's a hex value representing the flags? so 0x1020 (hex) is 4128 (dec) meaning ARMING_DISABLED_BOXFAILSAFE (4), ARMING_DISABLED_FAILSAFE (1), ARMING_DISABLED_RX_FAILSAFE (2), ARMING_DISABLED_NOPREARM (8)

Is that the right way to do this?

DanNixon, 2017-08-28:
You'd need to convert the hexadecimal value to binary, each bit that is on is a flag. As written in the Wiki you'd then need to look at the correct version of `runtime_config.h` to find out what flag each bit corresponds to.

For example, 0x1020 (bits 5 and 12) for 3.2 RC4 is throttle too high and in CLI.

DanNixon, 2017-08-28:
@ABLomas 0x1004 is CLI and RX fault, 0x1044 is CLI, RX fault and not level.

All of your fault codes for 3.2 RC4 point to a faulty or misconfigured RX, can you confirm it is working correctly in RC4.

ABLomas, 2017-08-29:
I do not see any issues with RX on "Receiver" tab.
RSSI (AUX4) is a bit high (2040), but this is not configurable (it's going over 2000 a bit when TX is close to RX), can't see any other issues with RX - channel values move, when i move sticks, directions are OK.
Works fine with 3.1.7, don't know where to get RC3/RC2 binary to verify, dead in RC4.

DanNixon, 2017-08-29:
So your RX is dead in RC4? What RX is it?

ABLomas, 2017-08-29:
No, RX is working. But FC does not arm ;-)
RX is OpenLRS micro. There is known issue with SBUS scaling on OLRS RX - endpoints are a bit out of range (smth like 920...2080 at max), easy to fix on TX, but RSSI endpoints aren't configurable, so they are in smth like 960...2060 range.
No issues on earlier firmware (many boards tested and still used with OLRS RX), maybe something changed in RC3/RC4 and even slight out-of range value disables RX?

Thanks for description how to decrypt arming disable flags - i will test with another RX and write back.

ABLomas, 2017-08-29:
Just to add:
- default failsafe range is way outside any channel range, so RSSI at min/max does not trigger failsafe
- i used 50ohm dummy loads on TX and RX, this causes RSSI to drop. So, when all channels are in 1000....2000 range it refuses to arm anyway. I guess, RSSI (AUX4 - CH8) with 2040-2080 value is not root cause.
RX is not so hard to replace, but it is working with earlier releases, so hm. I still have few frsky rx'es around and will try to replace it at evening.

mikeller, 2017-08-29:
How is your TX connected to your controller? The current SBus implementation in openLRSng has serious issues, if you can I'd switch over to mutliprotocol on the TX side, and SUMD on the RX side.

ABLomas, 2017-08-29:
VCC, GND and Signal - 3 wires.
"Signal" in this case is SBUS output from OLRS; PPM input on TX side.
As a long-time OLRS user (have ~20x RX'es) i do not know about any "serious issues" on SBUS side, except endpoint scaling (which has own workarounds). Few RX'es are still working on older FW (on other copters).
Can you clarify, please, which exactly "serious issues" you have in mind? Something really wrong with SBUS, which had workaround on earlier versions, but this was removed in latest version of betaflight?

mikeller, 2017-08-29:
> "Signal" in this case is SBUS output from OLRS; PPM input on TX side.

That won't work properly, since the scaling between PPM / openLRSng channel encoding and the current SBus implementation isn't even linear.

If you are using PPM on the TX side, try using SUMD on the RX side, and you'll get a properly scaled and linear signal.

To get an idea of the issues, have a look at the changes that I am proposing in 
 openLRSng/openLRSng#123 in order to make SBus on openLRSng scale properly with non-Sbus protocols.

ABLomas, 2017-08-29:
> That won't work properly, since the scaling between PPM / openLRSng channel encoding and the current SBus implementation isn't even linear.

We are going offtopic here.
I can tell you that such setup:

1. was working before without noticeable issues
2. is still working on few copters

Scaling is easy to overcome by changing endpoints. RX properly sends 1000....2000 on all channels except RSSI (but as i posted earlier - even with RSSI within 1000....2000 range FC does not arm).

DanNixon, 2017-08-29:
@ABLomas 3.2 (specifically RC4) added additional sanity checks to ensure that the received RX signal is valid before it will allow arming. One of which is the RX indicating a failsafe condition, which is present in some of the fault codes you provided (see https://github.com/betaflight/betaflight/issues/3994#issuecomment-325491106).

My guess would be an issue with openLRSng given it is the only common factor to this issue. (I'd test this myself but I'm in the middle of moving house and all my stuff is packed ATM)

Interesting you mention it is working on some copters, can you provide some details about those? (BF firmware, FC board, RX, etc.)

ABLomas, 2017-08-29:
> @ABLomas 3.2 (specifically RC4) added additional sanity checks to ensure that the received RX signal is valid before it will allow arming.
Thanks, this explains many things and probably my guess about looptime was incorrect.
I still need to test this with other RX....

> Interesting you mention it is working on some copters, can you provide some details about those? (BF firmware, FC board, RX, etc.)
Several copters (usually for "long range" (in copter-terms - 2-4km away from pilot)) using the same tardanis => PPM => Deluxe 1W TX ===> MicroRX => SBUS => FC
FC's used: DYS F4, Omnibus F4 (non-pro) x2, Matek F4, Kakute F4 and this one with ancient SRF3
Firmwares: IIRC all are on 3.1.7 or slightly older version. This one was first to be upgraded to 3.2 - first i took build from jenkins (before RC1), it was working but had issues with filtering; upgraded to RC2, issues with vibrations (unrelated), upgraded to RC3 and immediately to RC4 - and found that i can't arm copter.... 
A bit confusing, because initially i changed looptime and it was working for some time, later reduced looptime and... You know story =)
So, i will test with different RX (and if it works - i will need to investigate possiblity to migrate to OpenTX 2.2 + MULTI protocol + change output proto on RX - lots and lots of work and issue solving).
Maybe you know exact change number which added some (strange, i would say) checks for RX in RC4?
Thanks

DanNixon, 2017-08-29:
#3826 and #3681.

Although there is nothing "strange" about what these checks do. Why would you want to allow the craft to arm if the RX is indicating a failsafe condition?

ABLomas, 2017-08-29:
Hmmm, AFAIK failsafe on SBUS is triggered by separate flag in SBUS packet.
Checking RX output i see that this flag isn't set (and when it's set (TX off) GUI correctly indicates parashute icon on top). Channel values are inside "Failsafe trigger" range.
I do not know how else RX can indicate failsafe condition if:
1. it sends packets with all 16 channel values
2. failsafe flag isn't set
3. channel values are bigger than minimum for failsafe trigger and less than maximum.

Going to read about those PR's, thanks.

DanNixon, 2017-08-29:
I'm not suggesting any of these are a cause, but these (RX related) conditions also prevent arming:
- sending malformed packets
- timeout
- bad initial channel values (although this should trigger BAD_RX_RECOVERY, rather than RX_FAILSAFE)

mikeller, 2017-08-29:
@ABLomas: A lot of what you say above is based on the assumption that the encoding of channels in openLRSng is identical for all protocols. This is wrong, since the current implementation of SBus for openLRSng is a hack that does not take into account the compression that openLRSng does for the ends of the range on all other protocols, but simply bit-shifts the SBus values to make them fit into the openLRSng frame.

For this reason, it is not recommended to use a combination of SBus and any other protocol in openLRSng, and adjusting the endpoints will only stop it from triggering failsafe immediately, but not make it work properly.

ABLomas, 2017-08-29:
You are probably right.
But hey, moving away from current configuration is not easy task...
Deluxe has no working binary with MULTI input (if someone still sits in #openlrs at freenode - you can check about month back in history when we tried various binaries, etc - something wrong with serial), so only way is to go with SUMD on receiver. Not sure if this will fix things, but i check.

Anyway, i know at least one pilot nearby using similar setup and i think there are many in a world. Worth at least one BOLD line in release notes (maybe final release notes) that their old OLRS+SBUS setup won't work with 3.2, as it was working before.
Title on this issue is completely off ;-) Should i change it (so others could possibly find it), or just finish tests with SUMD and stuff and post together with workaround?

DanNixon, 2017-08-29:
It would be good to mention the RX in the title, it seems for whatever reason to be related to the issues you are seeing.

My FrSky SBUS receivers work fine with the changes made in 3.2.

mikeller, 2017-08-29:
@ABLomas:

Yes, changing configurations requires effort, and time. I had to wait for over a year until somebody implemented the features I wanted for openLRSng (serialRX and telemetry on the same port, digital RC data on the TX, configuration of the TX through the controller). And even then that person was myself in the end.

I am happy to work on getting multiprotocol to work with the Deluxe module - it is different from the other modules as in it has two serial ports (1 of them USB), and I don't have the hardware, so I could not test when I implemented multiprotocol support for openLRSng. (I thought @DTFUHF had tested multiprotocol with the deluxe module at some point, but looks like I am wrong.)

If you want to work on this, please join the Betaflight slack on http://www.betaflight.ch/, and we can discuss / debug there.

Re adding a warning, I think it is more openLRSng that should come with a warning that combining SBus and other protocols does not work - when doing so, the edges of the RX range get spread out, causing the signal to drop out of the valid range for very little change when it is around min / max.

ABLomas, 2017-08-31:
Well, let's look from users perspective.
This copter is older than mini-H, started with ancient FreeFlight FC (pre-NAZE32), lived through various OpenLRS versions and hardware (including initial random stuff before ctn and kha took OLRS into next level). So, several years and easily 1000+ km flown on OLRS, using PPM/SBUS. I never had issues with failsafe triggering and protocols (no, OLRS is not perfect - range issues, brownouts and other stuff<...> but not related to FC<==>RX communication).
So, outlived many baseflight versions, migrated to betaflight, lived up to 3.2 and boom - suddenly RX which was working so many years became incompatible =)
It seems that users have no easy workaround. SUMD has no direct output on most RX'es, so need to solder wire directly to transistor input, on SMD resistor. This will be a problem for many (and even so, i still can't get SUMD working on latest BF - i see signal on scope, but... But this may be my fault, i spend not much time debugging stuff).
What works - PPM. But this require re-wiring on RX side AND on FC side, which (usually) means you need to disassemble copter. And also get few other limitations (failsafe set not by flag, but by channel values, latency and so on).
So, purely from users perspective - just add one more CLI variable (you can even borrow it from INAV - "nav_extra_arming_safety=OFF") which would let users skip those extra safety checks and allow to fly at own risk (esp. having in mind so much time/mileage flown with this RX - i believe that it's working perfectly, compared to BF =) Also, you do not tested many, many other RX'es - there are so many exotic setups around, some made on LoRa modules, some on random stuff using arduinos - so while they are working now - they will prob. stop working after release and again, you will force all users to "fix" their RX'es (not sure what to fix on mine, heh)?
So, fixing SBUS output (i think, there is changosurf patch for SBUS scaling, but not in mainline - not sure if it will fix everything) does not mean that all RX'es will be compatible.
I'm not using slack, but will try to install and join channel later, but **for now simple workaround for users - forget DYNAMIC_FILTER, revert back to 3.1.7**....

mikeller, 2017-08-31:
@ABLomas:

> SUMD has no direct output on most RX'es, so need to solder wire directly to transistor input, on SMD resistor. This will be a problem for many (and even so, i still can't get SUMD working on latest BF

I think there is a misunderstanding. SBus and SUMD use the same pin on the RX: the serial TX pin. The same applies for the flight controller - the signal has to go to a pin that is capable of serial RX. The switch over from SBus to SUMD is hands-off configuration only, and - once the flight controller and TX module are hooked up to USB - can be performed in under 2 minutes (I've done it a lot when developing and flight testing the patch for SBus scaling):

1.) in the openLRSng configurator, RX tab, switch the assignment for the serial TX pin from 'SBus' to 'SUMD';
2.) in the Betaflight configurator, Configuration tab, switch the serial RX protocol from 'SBus' to 'SUMD';
(3.) if your flight controller has a switchable inverter on the UART you are using, go into CLI, and execute `set tlm_inverted = off`, and `save` to switch it off;
4.) Done.

ABLomas, 2017-09-01:
This topic is already mess - i started with looptime, continued with SBUS standarts and now debugging SUMD ;-)
Anyway, back to topic - no, there is no simple solution for users.
For F4 users - still need to un-invert SBUS
F3 users - dunno, but SUMD is not working on my FC anyway. I see fine SUMD signal coming out of FC, so it should work on FC "by default" (tlm_inverted is OFF by default), but not working on FC side (bars on RX tabs aren't moving, does not matter sbus or tlm inversion is turned on or off):
[image]
For now i attached another LRS as workaround (QLRS), working on desk, but not sure how it will work in field. Anyway, i think "strict-rx-control-before-arming" would be much better for end-user.

mikeller, 2017-09-03:
> For F4 users - still need to un-invert SBUS or re-wire to un-inverted port.

I do not quite understand how you come to assume this? The serial port on the openLRSng RX module does not support inversion, so the RX signal is uninverted - no matter if you select SBUS or SUMD. Hence, if your RX module is connected to an input of the flight controller that can deal with the uninverted SBUS signal, it will be able to deal with the uninverted SUMD signal - on all MCU types. 

Please disregard what I said about `tlm_inverted`, I had a senior moment, since `tlm_inverted` only applies to telemetry.

ABLomas, 2017-09-13:
In short:
 - i managed to get SUMD working on OMNIBUS F4 V2 (without software inverter), using the same port as SBUS, without any modifications (re-tested with 3x more RX+FC pairs) - requires only minor software re-configuration on RX/FC side
 - SUMD still not working for me on spracing F3/DODO/Omnibus F3, using the same port as SBUS, using inversion on FC. Needs more time on that, as i'm using another set of OLRS RX'es, so still needs debugging. SBUS works fine (receiver tab, did not tested for arm'ing)
 - as i used F3 MCU on my copter, i still do not have working SUMD here and can't confirm if i can arm copter using SUMD. Will verify later and get back with information

I also see that @mikeller added this: https://github.com/openLRSng/openLRSng/commit/7c54392b24fadbda93aa78cd6ce738b9c0ab8375
So, prob. latest firmware should also resolve SBUS incompatibility?

Anyway, please keep this open for some time, after vacation i will test all combinations and at least post workaround, until 3.2 will be released.

mikeller, 2017-09-13:
@ABLomas: Thanks for testing, and good to hear that there is some success. I am a bit surprised that it isn't working on the F3, it almost sounds like there is an inverter involved somewhere, but as far as I know none of the openLRSng RX has an inverter on the serial output.

And yes, I have merged a change into openLRSng that does standardise the output of SBus to be identical to the output of the other protocols. I will look into doing a release with this shortly.

ABLomas, 2017-10-21:
OK, so finally found some time to do tests:

- SUMD works at least on BroTronics SubMicroLRS RX (as a workaround for initial issue)
- SUMD does not work on openLRSminiRX (https://github.com/openLRSng/openLRSminiRX). I have lots of them but at least on 3x tested - activating SUMD does nothing - FC does not recognize signal as SUMD. At least on scope it looks 1:1 like SBUS.
- Workaround for them - upload latest OpenLRS code with https://github.com/openLRSng/openLRSng/commit/7c54392b24fadbda93aa78cd6ce738b9c0ab8375 - then at least it's possible to use SBUS + 3.2.1 BF. Issue - can't set channels below 1005 - it re-sets immediately to ~1260 and that's all, so i scaled my channels to have endpoints 1005...1995.
Tested for few seconds in air so far (weather...) but everything else looks good.

So, to repeat - SUMD may work as option on some OLRS RX'es, where it does not work - get latest (1.8.5 at this moment) arduino IDE and build code for your RX, using latest OLRS zip from github. I hope https://github.com/openLRSng/openLRSng/issues/133 will get some attention and SBUS fix will be released as binary.

mikeller, 2017-10-22:
Thanks @ABLomas for testing. 'Make openLRSng release' is on my to-do list.

haslinghuis, 2022-05-08:
Closing as inactive

closed 2022-05-08 by haslinghuis