betaflight/betaflight #3998: dronin re1 v2 fc voltage problem

andro46, 2017-08-29:
Hello.
I have problems with voltage output on betaflight and osd. It show 3,2V max when go to voltage scale and make it to max 255.. if i have lower numbers the voltage is even lower. Have some one any idea
what to do?
[image]
[image]

robert-hoffmann, 2017-08-29:
Start by setting your max/min/warning voltage
Mine; max (4.3)/min (3.3)/warning(3.5)

andro46, 2017-08-29:
i have tryed.. whats the point if the max shows only 3,2 when i turn on the drone

robert-hoffmann, 2017-08-29:
You have to tell it what the min max values of your battery are, otherwise it will never work anyways.

IE. if i set max to 4.2, and actual voltage is 4.21, the system freaks out and i get crazy values. Configure min/max, then play with scale ..and if still messed up, then explain whats wrong

andro46, 2017-08-29:
i have tryed with normal cell voltage, max 4,2 min 3.3 .... and then tryed to correct with scale. and always the same problem. I cant make it higher than 3.2 V. I use 4s batterys

robert-hoffmann, 2017-08-29:
Max is 4.3 on a 4s ..if you say max is 4.2 and the battery is actually 4.22 ..the system freaks out....

andro46, 2017-08-29:
[image]

andro46, 2017-08-29:
ok now i have 4.3.. nothing ;/

andro46, 2017-09-01:
still didnt solve the problem.. any one has a other idea?

mikeller, 2017-09-03:
Your firmware version is quite old, maybe try with 3.2.0-RC4.

closed 2017-10-23 by jflyper