betaflight/betaflight #4001: BF 3.2 DSHOT1200 and OmnibusF4 makes motor 1 twitch

ralenekov, 2017-08-29:
My setup: Soft-mounted Omnibus F4 (MPU-6000), 8/8, PT1, dynamic filter, zeroed notches, DSHOT1200, Panasonic FM 1000uF/35V, Airbot Wraith 35A running BLHeli32 32.1

Betaflight 3.2-RC3/RC4

With props removed and quad on a soft surface:
1. Selected DSHOT1200, using BLHeliSuite32 => every motor is smooth.
2. Selected Multishot => Using control from the radio, every motor is smooth, arm/pitch/roll/yaw.
3. Selected DSHOT600 => Using control from the radio, every motor is smooth, arm/pitch/roll/yaw.
4. Selected DSHOT1200 => Using control from the radio, every motor is smooth on every command except left roll. On left roll motor 1 (not sure about 2) start to screech and twitch.

[image]

[image]

Log (DSHOT1200/DSHOT600/Multishot): [BTFL_BLACKBOX_LOG_20170828_232918.zip](https://github.com/betaflight/betaflight/files/1259973/BTFL_BLACKBOX_LOG_20170828_232918.zip)
Config: [diff 3.2-rc3.zip](https://github.com/betaflight/betaflight/files/1259974/diff.3.2-rc3.zip)

emski0, 2017-08-30:
i don't know if it's the same problem, but i had 3 motors twitshing on Dshot1200( actually all dshot protocols).Using Omnibus F4SD on 3.2 RC3 and bl-heli32. ( All of this was using the motor tabs.)
On multishot no twitches.
I found out that the beeper was causing the twitches. I had  my failsafe set to beep. Once i remove the beeper mode from failsafe , no more twitches, and flies smoothly now.
Give it a try.

coreysnyder, 2017-08-31:
I don't think Omnibus F4 is compatible w/ DSHOT1200.

ralenekov, 2017-08-31:
Update:
Tried 3.1.7, only rx and DSHOT1200 configured - still reproducible. DSHOT600 is fine.
Tried 3.2-rc4 again. Deactivated all buzzer functions, all features, lowered PIDs - still reproducible. DSHOT600 is fine.

It happens if it is a **clean ROLL command** (left or right). If it is mixed with pitch, the twitching doesn't appear.

It happens in flight as well. This is from an old log. Please take a look at the command:
[image]

I will will stay away from DSHOT1200 on this target for now...

jmc007, 2017-09-01:
Same here with F4 omnibus, Spedix GS30 BLHeli 32, Emax 2306. Motor 3 knocking noise, stuttering. Same on an F3 Omnibus board.
Switching to DShot 600 as you suggested was the only cure for me. Thanks again.

So is it confirmed what coreysnyder said? That the Omni boards have a problem running DShot 1200?

etracer65, 2017-09-01:
Have you tried:

`set sdcard_dma = OFF`

Not sure but it sounds similar to other DSHOT problems and other dma conflicts causing motor twitches and other problems.

jmc007, 2017-09-02:
Just tried it still juddering spluttering on motor 3 with DSHOT1200.

Cheers anyhow.

mlilasic, 2017-09-04:
Had similar issues but on F3 FC. Here are details:

Had to get customized bf fw build and move motors 3 and 4 to 5 and 6 motor outputs.

Here are my findings:

Issue:
[url]https://www.rcgroups.com/forums/showpost.php?p=38006423&postcount=1222[/url]

Issue solved:
[url]https://www.rcgroups.com/forums/showpost.php?p=38072934&postcount=1304[/url]

[betaflight_3.2.0_X_RACERSPI_DSHOT1200_timer_fix.hex.zip](https://github.com/betaflight/betaflight/files/1274297/betaflight_3.2.0_X_RACERSPI_DSHOT1200_timer_fix.hex.zip)

adamtfc87, 2017-09-05:
Hello! I'm currently using the EFAW F4 flight controller which runs off of the vgooddhf4 firmware (using luxF4 for 3.2 RC4) and having issues with dshot1200 similar to what was posted above. Everything seems to work wonderfully with dshot600 similar to others. In flight there is horrible bounce back and desyncs, also motor twitching. Can you guys help me get a working firmware so I can run dshot1200 please? Thank you very much

jmc007, 2017-09-06:
Copy and pasting what I just posted on the BLHeli 32 thread on RCGroups.

"I got DSHOT 1200 running without any stutter on motor 1 now with these Spedix GS30's.

I noticed this written in the Wraith32 35a and the 25a specs:- "Dshot up to at least Dshot1200, and at rates up to at least 32kHz is now supported."

Turned mine down from 48khz and honky dory. Not had a test flight yet, though a quick hover and none of the horrible stutter judder on motor 1. Happy days. "

adamtfc87, 2017-09-09:
Found out my issue, defective escs. Went through 4 bad ones out of the 5 I purchases before having a good set of 4 working. It was the voltage reg going out, and airbot is sending me replacements because it is becoming an issue aparrently. Problem solved!

fixmania, 2017-09-11:
I have too. I use Omnibus F4 (max pid loop is 8K) and Airbot 35A Dshot-1200. Also motor 1 has a problem as like above. When in flight, occur desyncs and flip of death every flight. I tried change motors and escs, but same result. I think only BF 3.2 firmware has a problem. So hope to solve this issue.

rustbg, 2017-09-17:
Same here.
Everything is ok, except DSHOT1200.
Motor 1 acts strange. 
Omnibus F4 Corner with Wraith32 mini 25A.

peecao, 2017-11-18:
OmnibusF4 is definitely not compatible with DSHOT1200.  
I HAVE TRIED EVERYTHING.  I bring it back to DSHOT 600 or below and everything works perfectly.

etracer65, 2017-11-18:
This was fixed on BF 3.2.1.  Current release is 3.2.2.

closed 2017-12-20 by ralenekov