betaflight/betaflight #4003: Amperage Meter

ChimpsterFPV, 2017-08-29:
Hi,
today i tryed to calibrate the Amperemeter on my BFF3 Board with 3.2.

I can Calibrate the Ampere Meter pretty perfect on 3.1.7, the OSD says that 1204mA was used and my ISDT Chrger loads about 1250mA in the Battery which is very good for me.

The Problem on 3.2 is, that it´s not able to set the value "Offset in milliampere steps" to a negative Value like in 3.1.7  . The Offset should be a negative Value im my case, then it could be perfect.

Here is my Setup:

name CHIMP
resource BEEPER 1 C15
resource MOTOR 1 A06
resource MOTOR 2 A07
resource MOTOR 3 B08
resource MOTOR 4 B09
resource MOTOR 5 B00
resource MOTOR 6 B01
resource MOTOR 7 A03
resource MOTOR 8 A02
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B07
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 B01
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 B00
resource SERIAL_RX 12 NONE
resource LED 1 NONE
resource LED 2 NONE
resource LED 3 NONE
resource RX_BIND 1 NONE
resource RX_BIND_PLUG 1 NONE
resource SPI_SCK 1 B03
resource SPI_SCK 2 B13
resource SPI_SCK 3 NONE
resource SPI_MISO 1 B04
[... 472 lines cut ...]
set d_level = 75
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_sensitivity = 55
set level_limit = 55
set horizon_tilt_effect = 75
set horizon_tilt_expert_mode = OFF
rateprofile 0

set rc_rate = 0
set rc_rate_yaw = 0
set rc_expo = 0
set rc_expo_yaw = 0
set thr_mid = 45
set thr_expo = 32
set roll_srate = 0
set pitch_srate = 0
set yaw_srate = 0
set tpa_rate = 5
set tpa_breakpoint = 1650

Thanks for your Support
Realy appreciate that!!!
Erkan

ofiratt, 2017-09-26:
I have the same issue, cant calibrate the current sensor correctly, usually it easy to calibrate with two variables equation. **but scale cant be decimal**

There are a few places in the code that need to be fixed
  power.html: 
     in the code you can find `parseInt($('input[name="amperagescale-' + index + '"]').val());`
     but it need to be  `parseFloat($('input[name="amperagescale-' + index + '"]').val());`
  MSHelper.js:
   in ` function send_next_current_config()` you use `.push16(CURRENT_METER_CONFIGS[configIndex].scale)`  but push16 cant set decimal number and it convert it to zero

mikeller, 2017-09-26:
@ChimpsterFPV: The negative values issue was fixed in betaflight/betaflight-configurator#651. It will be included in the next configurator update.

@ofiratt: Betaflight uses integer to store / communicate configuration values, for various reasons. This usually works fine, and the value range available when using int16 seems to be enough for all configuration settings, when scaled properly. If you specify what int16 range you'd like to see for the scale, we can look into implementing it.