betaflight/betaflight #4005: BF 3.2 RC4 - OSD Default RSSI Alarm

bigd942, 2017-08-29:
The default setting for the RSSI on the OSD is 20, on the Taranis the default is 45, would 35 or 40 not be a safer default to allow some safety?

mikeller, 2017-08-29:
We definitely could change it. My suspicion is that, dependent on the TX / RX you are using, either 40% or 20% might be a good choice, or probably none of these.

What do others in here think the default value should be?

Arakon, 2017-08-30:
As far as I know, the Taranis doesn't use a percentage value, rather a dB one or something.. so 50 on the taranis is very different from 50% in OSD. My XM receivers for example drop to 55% fast even just being behind trees 30m away, despite having 200+ meters of range. The one time I did have an actual failsafe with one of them, my RSSI dropped below 25% before it happened, so I think the 20%  warning value may be fairly close already.