betaflight/betaflight #4008: 2 Problems with SPRACINGF3

BASSBEAT4LIFE, 2017-08-30:
1) When i enable the option MOTOR STOP the motors still start spinning when armed.... ? 
2) I'm not able to get any voltage information of the battery so i can't set a min voltage of cut off... (1S battery)

best regards

jflyper, 2017-08-30:
@BASSBEAT4LIFE 
You seem to have configuration problems. Please redirect your questions to the following RCG thread. Thank you in advance.

https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread

closed 2017-08-30 by jflyper

richard-scott, 2017-08-31:
@BASSBEAT4LIFE If you have Airmode enabled your motors will always spin when armed.
Source: https://github.com/betaflight/betaflight/wiki/Frequently-Asked-Questions#how-do-i-enable-air-mode-