betaflight/betaflight #4010: Small possible performance improvement

henthe-5, 2017-08-30:
Just a thought when checking out sensor calculations.

Line 239 in flight/imu.c uses floating-point division to calculate the inverse square root for normalization of measurements.

Maybe could be faster by using fast inverse square root for this as in 
https://en.wikipedia.org/wiki/Fast_inverse_square_root

for example:

float Madgwick::invSqrt(float x) {
  float halfx = 0.5f * x;
  float y = x;
  long i = *(long*)&y;
  i = 0x5f3759df - (i>>1);
  y = *(float*)&i;
  y = y * (1.5f - (halfx * y * y));
  y = y * (1.5f - (halfx * y * y));
  return y;
}

DieHertz, 2017-08-30:
GCC is good at such optimization by itself, at least for X86-64 it easily generates code for reciprocal square root without FP divisions.

wind0r, 2017-08-30:
I first thought the compiler with the fast math option would already do similar, but i just did some checking/counting without a coffee, with looks like the change would be faster.
I think we should do test this advise some more. 

This may all is incorrect or totaly wrong. 
Godbolt.org with arm 5.4.1 -Ofast -mthumb -mcpu=cortex-m4 -march=armv7e-m -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -Wdouble-promotion and arm 6.3 and -Ofast -mthumb -mcpu=cortex-m4 -march=armv7e-m -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -fsingle-precision-constant -Wdouble-promotion

testcode:
```
#include <math.h>

float rsqrt( float number )
{
	long i;
	float x2, y;
	const float threehalfs = 1.5F;

	x2 = number * 0.5F;
	y  = number;
	i  = * ( long * ) &y;                 
	i  = 0x5f3759df - ( i >> 1 );
	y  = * ( float * ) &i;
	y  = y * ( threehalfs - ( x2 * y * y ) );

	return y;
}

int next_sqrtf2(float x)
{
    return (1.0f / sqrtf(x));   
}

int next_number3(float x)
{
    return rsqrt(x);
}
```
asm:

```
next_sqrtf(float):          sum: 33 cycles
  vmov s15, r0              1 cycle
  vsqrt.f32 s14, s15        14 cycles
  vmov.f32 s13, #1.0e+0     1 cycle
  vdiv.f32 s15, s13, s14    14 cycles
  vcvt.s32.f32 s15, s15     1 cycle
  vmov r0, s15 @ int        1 cycle
  bx lr                     1 cycle

next_rsqrt(float):          sum: 15 cycles
  ldr r3, .L6               1 cycle
  vmov s15, r0              1 cycle
  sub r3, r3, r0, asr #1    1 cycle
  vmov.f32 s13, #5.0e-1     1 cycle
  vmov s14, r3 @ int        1 cycle
  vmul.f32 s13, s15, s13    1 cycle
  vmov.f32 s15, #1.5e+0     1 cycle
  vmul.f32 s13, s13, s14    1 cycle
  vfms.f32 s15, s14, s13    3 cycles
  vmul.f32 s15, s15, s14    1 cycle
  vcvt.s32.f32 s15, s15     1 cycle
  vmov r0, s15 @ int        1 cycle
  bx lr                     1 cycle
```

so it looks like it maybe is worth the effort and testing.
I didnt checked the math, result or code. i only counted the cycles.

gke, 2017-09-08:
Just an observation that there are quite a few divisions on the code mainly in initialisation. For example /10.0 which should * 0.1 automatically. Some languages do not permit compiler transformations to reciprocal. Probably I should check for C ;).

DieHertz, 2017-09-08:
@gke We have `-ffastmath` which is there specifically for these kinds of optimizations.
It's not languages but more how strictly IEEE-754 conforming the compiler is.

closed 2018-04-21 by DieHertz