betaflight/betaflight #4017: BF 3.2.0 setpointweight is disabled by defaul label:BUG

jirif, 2017-08-30:
In betaflight 3.2 is Setpointweight disabled by default settings. Propably mistake in commit https://github.com/betaflight/betaflight/commit/45c578c8b65ac064aaf0ccdf833c49cc1c506479 ? When relax ratio is 100 whish is default, setpont weight is getting ignored. Maybe this simple correction  was idea of the original author ...
```
            float dynC = 0;
-           if ( (pidProfile->setpointRelaxRatio < 100) && (!flightModeFlags) ) {
+           if ( (pidProfile->dtermSetpointWeight > 0) && (!flightModeFlags) ) {
                dynC = dtermSetpointWeight * MIN(getRcDeflectionAbs(axis) * relaxFactor, 1.0f);
            }
 
```

borisbstyle, 2017-09-07:
100 is actually not doing nothing (disabled) so it means it has no point to do anything

closed 2017-09-07 by borisbstyle

jirif, 2017-09-07:
```
             float dynC = 0;
            if ( (pidProfile->setpointRelaxRatio < 100) && (!flightModeFlags) ) {
                dynC = dtermSetpointWeight * MIN(getRcDeflectionAbs(axis) * relaxFactor, 1.0f);
            }
            const float rD = dynC * currentPidSetpoint - gyroRateFiltered;    // cr - y
```
dynC stay at zero when setpointRelaxRatio == 100
float dynC = 0;
instead of correct
dynC = (pidProfile->dtermSetpointWeight / 127.0f) * 1 = **0.4724409448818898 * 1**
as we have before :)

so 
```
         const float rD = dynC * currentPidSetpoint - gyroRateFiltered;    // cr - y
```
is 
rD = 0 * currentPidSetpoint - gyroRateFiltered = **0  * currentPidSetpoint - gyroRateFiltered**
instead of correct
**rD = 0.4724409448818898  * currentPidSetpoint - gyroRateFiltered**
zero * something is still zero, right ?

Thats all, have look at your commit https://github.com/betaflight/betaflight/commit/45c578c8b65ac064aaf0ccdf833c49cc1c506479
After that change dtermsetpoint weight working only when setpointrelaxratio is set <100 (in gui <1)

If is supposed to disable setpointweight parameter by setting relaxratio to 1 it should be noted because its confusing for people using setpoint weight feature with relaxratio 1 ...

PR #4076 (2017-09-08): BF 3.2 setpointweight is disabled by default
fix #4017