betaflight/betaflight #4020: Implement DSHOT using TIM_DMAR

ledvinap, 2017-08-30:
It is possible to update TIM channels using TIM_DMAR.
This will greatly reduce number of required DMA streams necessary - one per timer instead of one per channel.
It will also allow much greater configurability - with CCR double-buffering, it should be possible to use any channel to trigger DMA (actual value is used only after timer update event) or update event (but previous value would be used on update, making implementation slightly more complicated).

This may also help with DMA 'collisions', improving stability. Also update will be more deterministic (all channels updated sequentially, avoiding concurrent, data dependent triggering of multiple DMA transfers).

Any volunteer to look at this ? ;-)

jflyper, 2017-08-31:
Interesting!!!

sambas, 2017-08-31:
ProShot could be done so that all four timer channels are updated with one dma in 16 burst mode (4x4 datablocks). CCR values would need to arranged in memory as [CCR1_1, CCR2_1, CCR3_1, CCR4_1, CCR1_2, CCR2_2, CCR3_2, CCR4_2, CCR1_3, CCR2_3, CCR3_3, CCR4_3, CCR1_4, CCR2_4, CCR3_4, CCR4_4]

ledvinap, 2017-08-31:
@sambas : This is the same for DShot. 

Just allocate enough memory for all consecutive channels on given timer (first register+count is specified for DMAR, dummy memory is necessary when channels are not consecutive) and add info about offset+stride to motor info.

sambas, 2017-08-31:
proof of concept, Single timer (TIM8) all four outputs enabled and outputting array of 68 words (4*16 data+4 leading zero) with TIM8_UP DMA channel. All channels synced nicely.

[image]
[image]

DieHertz, 2017-08-31:
This is so cool, @ledvinap, how did you come up with this idea?

DieHertz, 2017-08-31:
@ledvinap I tried googling `TIM_DMAR`, but didn't find anything. Are you referring to `DMAR` register?

jflyper, 2017-08-31:
I didn't notice they are there, but available for selected timers on all STM32 family.
Search with TIMx_DMAR, but there's very little example/mention about it.

sambas, 2017-09-02:
I'm playing with code here https://github.com/sambas/cleanflight/commits/timup
I have multiple timers working now,

[image]

ledvinap, 2017-09-05:
@sambas : Great work. Can you please create PR for your code? It's much easier to comment etc ...

closed 2018-04-21 by DieHertz