betaflight/betaflight #4022: Smartport and smartaudio not working on omnibus f4 pro corner (omnibusf4sd)

miniclip12, 2017-08-31:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://rcgroups.com/forums/showthread.php?t=2464844), [Slack](https://slack.betaflight.tech) or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1.) Describe your problem;
2.) Include ways to reproduce the problem;
3.) Provide information about your flightcontroller and other components including how they are connected/wired;
4.) Add the used configuration and firmware version;
5.) Create a `diff` and post it here in a code block. Put \`\`\` (three backticks) at the start and end of the `diff` block; and
6.) Remove this Text :).
The smartaudio does not work on any of my omnibus boards now with rc4. The smart port telemetry doesn't work on any of the omnibus boards regardless with xsr. I have uninverted the telemetry and tried many uarts including soft serial. Currently on the f4 pro corner I am using uart 1 rx for blheli 32 telemetry, uart 3 rx for sbus, uart 6 tx for smartaudio, and softserial 1 for smartport. I have switched the uarts around and nothing is working for the smartaudio and smartport. I cannot test the blheli 32 each telemetry because smartport is not working. Thanks in advance

mikeller, 2017-08-31:
@miniclip12:

1. Read the instructions above on what information is required from you for others to be able to help you;
2. Add the information in a comment on this issue.

jflyper, 2017-08-31:
@miniclip12 

This is a configuration issue. I have posted a comment in the RCG thread mentioned below.

https://www.rcgroups.com/forums/showthread.php?2801694-Omnibus-f4-pro

closed 2017-08-31 by jflyper