betaflight/betaflight #4025: Modify Operational models via CLI

sorhawell, 2017-08-31:
My first issue posting, I hope it fits the standard

It seems the operational models 'race' or 'freestyle' only can be set via osd menu. Wouldn't be possible to expose these configurations to the CLI interface?

For a FC with no OSD it is not possible to activate operational models 'race' and then lua script cannot turn pitmode on/off 

https://github.com/betaflight/betaflight/wiki/Unify-Smartaudio#operational-models

jflyper, 2017-08-31:
@sorhawell 
> It seems the operational models 'race' or 'freestyle' only can be set via osd menu. Wouldn't be possible to expose these configurations to the CLI interface?

It is planned to have such interface. Will probably be implemented in 3.3 or later. You can also search the issue ticket by entering "vtx" keyword, and you will find lots of different request on various UI methods.

> For a FC with no OSD it is not possible to activate operational models 'race' and then lua script cannot turn pitmode on/off

I think this is a separate feature request...

sorhawell, 2017-09-04:
thx for reply i follow those threads

closed 2017-09-04 by sorhawell