betaflight/betaflight #4026: KAKUTE F4 and 3.2 smartport bug

beb38138, 2017-08-31:
Kakute f4
X4rsb
        After a fresh flash to 3.2, I will have smartport and smart audio, but after a few power cycles, i lose that function (although all settings are same) until i reflash and do full chip erase. If I stay on 3.1.7, smartport works as it should. This only happens with 3.2..
        _setup_ 
x4rsb connected to uart 1 (smartport pad), smart audio (IRC Tramp) to UART 6tx, dsot600, dys xsd30a
       _configuration_
bf configurator 3.2.1
KAKUTE 3.2

\\\# diff

# version
# Betaflight / KAKUTEF4 (KTV1) 3.2.0 Aug 28 2017 / 11:55:44 (b2cd7294e)

# name
name Big Rig

# resources

# mixer

# servo

# servo mix

# feature
feature AIRMODE

# beeper

# map

# serial
serial 0 32 115200 57600 0 115200
serial 5 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# master
set motor_pwm_protocol = DSHOT600
set pid_process_denom = 1

# profile
profile 0

# rateprofile
rateprofile 0

# \\\

mikeller, 2017-09-01:
Have you updated the firmware on your RX to latest? There is a bug in older firmware that causes it to lose SmartPort after some time.

beb38138, 2017-09-03:
I used this cli command to get get s.port to work on led pad. Appears to be a hardware problem, not with code. You can close this issue if you wish.

feature SOFTSERIAL
resource LED_STRIP 1 NONE
resource SERIAL_TX 11 A01
serial 30 32 115200 57600 0 115200
set tlm_inverted = OFF
set tlm_halfduplex = ON
save

jflyper, 2017-09-03:
May be related with #3695.

In #3695, I've suspected that bidir inverter has a design problem, but haven't heard from the user since then.

closed 2017-09-03 by mikeller

jamming, 2017-09-05:
Kakute F4's bidir inverter is designed with mosfet.  For some special XSR  receivers, who's  input capacitance are higher than the normal, it may cause the smartport waveform distorted. Change a resistor on the KakuteF4 from 4.53K to 2.2K, the waveform looks good.  That may be the reason.

mikeller, 2017-09-05:
Thanks for analysing this, @jamming. Is it possible that you add instructions on how to do this to the [wiki](https://github.com/betaflight/betaflight/wiki/Board---KAKUTEF4)? (It's probably preferred if this can be done with external, non-SMD components, as a lot of users will not be comfortable with soldering SMD.

jamming, 2017-09-05:
I have updated the wiki, see "FAQ&Known Issues"  section.

mikeller, 2017-09-05:
Thanks, that's very helpful!