betaflight/betaflight #4029: Feed Smart-Audio data to Spektrum Telemetry receiver

coreysnyder, 2017-08-31:
It looks like some [work has already been done](https://github.com/betaflight/betaflight/pull/1798) to provide telemetry data to the Spektrum satellite receiver. Currently on RF1 they have the ability to also control the VTX band/channel/power through their transmitter. Could this same thing be setup on Betaflight?

[image]

Cheers!

jflyper, 2017-09-01:
IIRC, ...
- There has been a PR, which proposed an MSP extension for VTX control to run over SRXL(?) protocol, but was lost in time.
- The author was told that BF already have some mechanisms for doing it, and should consider conforming to it.
- The author never responded...

Abancalari, 2017-09-18:
Another suggestion that could also solve this might be to incorporate the current CMS structure into the "TextGen" function of the Spektrum Telemetry Protocol. The CMS would need some compacting as the spektrum displays only have ~5-7 vertical lines but once implemented, you would just need to use stick commands to navigate and be looking down at your display. This doesn't really use the native VTX function on the radios but is probably more useful as it's takes care of that issue, is fully featured (all the channels are available and not just the ones spektrum decides) and will easily update every time there is an update (instead of having to update LUAs every time). Just a thought. 

https://www.rcgroups.com/forums/showthread.php?2776065-Spektrum-Serial-Telemetry-Receiver-%28SPM4649T%29-Interfacing

Dessection, 2017-09-18:
Spektrum have added interface doc here: https://github.com/SpektrumFPV/SpektrumDocumentation/blob/master/Telemetry/Spektrum%20VTX%20Interfacing.pdf

AndersHoglund, 2017-10-22:
I have started looking into this, PR coming. Prototype needs testing.....

Dessection, 2017-10-23:
I'll gladly help test this if you need.

AndersHoglund, 2017-10-23:
Good offer, thank you. Prototype is here to pull/fork/build: https://github.com/AndersHoglund/betaflight/tree/spektrum_vtx_control
I have a Tramp HV incomming any day now. So, no real tests done yet. Only some GDB sessions to check a few things. Only to find the Spktrum DX18g1 1.20 VTX control menus are a mess, almost nothing according to spec.... And then there are other activities and fixed ongoing on the VTX -end of the system. So we are caught between two fires really. Not much point in doing a PR until those are put out. Phew.

AndersHoglund, 2017-10-23:
There is a prototype build job as well: http://andwho.sytes.net:8080/job/AndWho_Betaflight_Spektrum_VTX_control_prototype/

HFMan, 2017-10-24:
@AndersHoglund, 

I have been testing your prototype- it's working well.  I did find a couple bugs (the Frequency Table for Band A and Band E need to be swapped when using the Tramp, and the Tramp has 100mW as one of its indexed power settings that you don't account for.)  Other than this, it seems to be working well.

Abancalari, 2017-10-24:
Just tested with a CLRacing F4 and Unify HV Race. Have the same issue as HFMan  where A and E are swapped. I also had it lock me out from changing channels for a bit but that may have been due to trying pit mode (Which i do not believe works). I couldn't replicate the issue the second time. I also noticed it puts the Vtx into user frequency mode which is fine really. Was able to go to any channel that wasn't locked out by tx and was also able to succesfully switch from 25mw to 200mw and back. Awesome work @AndersHoglund! Will be testing at our next race.

AndersHoglund, 2017-10-24:
Thanks for testing @Abancalari and @HFMan , I will look into the issues. The prototype seems mature enough for a PR, and more wide spread testing. But the lockup and pitmode issue is a blocker.

I would also like to remove all the built in "knowledge" on all VTX device specific settings, there are VTX device indepentant API for Pitmode and Frequency, but not for Power. Having output device translation (VTX power) tables in the input device is sooo ugly I can puke.  Will address that in the PR that did the other device independance stuff in vtx_common #4265 .

AndersHoglund, 2017-10-24:
On the issue on swapped A and E band. 
Band and channel is NOT used towards the VTX. The device independant Frequncy API is used. And I have checked using GDB that the frequencies displayed in the DX18g1 menues are the actual frequencies sent to the vtx_common API. If that frequency translates to some different band/channel is not a problem as long as it is the correct frequency. Band/channel indexes are/can  be brand specific. The Spektrum definitions might not match for example the internal vtx_common or the Tramp  defs. Frequency is not brand specific, it is absolute.
So the Q is, was the correct frequency set?

Dessection, 2017-10-24:
Have only tested with the Unify Race. But it changes between 25 and 200mv without any problem, and the same with the frequencies i have tried. They end up at the right ones. Thanks so much!

AndersHoglund, 2017-10-24:
Thanks for testing @Dessection.
Small change made to setting of frequency,  made sure band was set to USER first. Re-testing might be needed. 
Also a swap of band A and E in the translation table, even if it is not used.

Dessection, 2017-10-24:
Well, even tho the Airware Fw needs an update to open all channels, it seems to work just fine both with going to the selected channel, and changing between 25 and 200 still. Have not tested with the Pit / Race option. Will look into that later. (That is for the latest build #6)

AndersHoglund, 2017-10-24:
The prototype branch is now squash into one commit and PR #4434 is submitted. Lets continue discussions and tests there.

closed 2017-10-24 by AndersHoglund