betaflight/betaflight #4030: Change from DSHOT600 back to MULTISHOT and can't calibrate motors anymore.

richard-scott, 2017-08-31:
**TL;DR**

1. Flash with full chip erase.
2. Disconnect USB.
3. Connect USB and Battery.
4. Change ESC protocol to DSHOT600, click Save and Reboot.
5. Change ESC protocol to Oneshot125, click Save and Reboot.
6. Disconnect USB and Battery.
7. Connect only USB (no Battery).
8. In Motors tab slide Master control to top.
9 Connect Battery.
10. Motors will instantly run at full speed instead of calibrating.

Only fix is to wait 10 seconds between 6 and 7.

**Long version**

I have an original (not clone) SP F3 Evo with [LittleBee ESC's](http://www.hobbyrc.co.uk/little-bee-spring-40a-esc-with-blheli_s-dshot-2-6s) and no additional capacitors added by me (that may be important to say, you'll see why soon).

Anyway, I can't re-calibrate my ESC's after going back from Dshot600 to Oneshot125 (or any other non-dshot protocol compatible with my ESC's).

To reproduce try this:

Flash the latest build of the firmware from [ci.betaflight.tech](https://ci.betaflight.tech/job/Betaflight/lastSuccessfulBuild/) with full-chip erase selected. In this case it was Build 147. (I've just tried RC4, and that is the same too).

Disconnect from USB and reconnect to USB.

In BFC I re-map Motor 4 to Motor 5 as that's where its wired for DShot. Then I quit BFC and using BLHeli Configurator I flash my ESC's with “16.67RC [blheli-multishot]” and ensure "Programming by TX" is enabled.

I then Quit BLHeli Configurator and disconnect from USB and Battery.
Reconnect only to USB.

Back in BFC I check that Oneshot125 is set as it is the default ESC protocol. I then go to the Motors Tab and Calibration works fine during at time e.g in the Motors tab, I acknowledge the risk slider, set master to the max, and connect my battery. The ESC's all beep nicely, once the beeping stops I lower the Master slider and they finish calibration. I test the motors with the slider and they all work fine.

With the battery still connected I then change the ESC Protocol to DSHOT600 and after a reboot (with the correct DShot ESC Beeps) I then test my motors from the Motors Tab and they still run. Obviously I'm not calibrating DShot.

I then decide I want to re-calibrate my ESC's so change the protocol back to Oneshot125 (or Multishot/Oneshot42). After a reboot the ESC's beep the Oneshot125 tones so I disconnect both USB and battery and I reconnect USB only. 

In BFC I check the ESC protocol is correct at the CLI and it is ONESHOT125. I then go back to the Motors Tab to re-calibrate only to find when I click the risk slider, and move master to the max that when I then connect my battery my motors run at full power.

Initially I thought that flashing the FC again helped reset things, but it seems that I can only re-calibrate my ESC's if I disconnect from ALL power and count to 10. Once I wait 10 seconds and re-connect only to USB I can then calibrate my ESC's again.

Oh, and I've just tried with a 16.63 firmware on my ESC's thinking it could be that, but it is still the same no matter which dshot ESC firmware I am using.

It is odd, as the ESC's beep the correct Beeps for Oneshot125 (compared to Dshot600) so I don't understand what's going on.

mikeller, 2017-09-01:
Interesting. Looks like at least in the case of switching from Dshot to MultiShot, it takes a couple of seconds for the ESCs to properly detect a power loss, and do a cold start. Could be that this intentional, so that brown outs in flight do not cause the ESC to stop working until it is armed by an idle signal...

richard-scott, 2017-09-01:
Are the ESC's powered by USB? Duh, I guess they must be for the BLHeli Configurator to work.

Anyway, I also seem to remember having the same issue when changing ESC protocol without the battery connected and then attempting calibration (but my mind is a bit fuzzy on that one as it took a few hours to work out the details of this issue). 

I seem to rememebr that this also caused the issue with no battery:

Flash FC, Change ESC protocol from Oneshot125->Dshot600->Oneshot125, then Calibrate.

Also, the ESC's beep the correct tone on power up from battery depending on which Protocol I've selected so they should know which mode they are in.

mikeller, 2017-09-01:
> Are the ESC's powered by USB? Duh, I guess they must be for the BLHeli Configurator to work.

Some flight controllers are known to bleed back USB power onto the 5V rails, so in some cases 'yes'.

richard-scott, 2017-09-01:
Ok, I'll close this as it appears that it may not to be a Firmware issue after all.

closed 2017-09-01 by richard-scott