betaflight/betaflight #4032: SmartPort over SoftSerial broken on BLUEJAYF4

andrejpodzimek, 2017-09-01:
As discussed on rcgroups ([from this post on](https://www.rcgroups.com/forums/showthread.php?2593106-BlueJayF4-STM32F4-based-FC/page68#post38017641)), I use SmartPort over SoftSerial on `BLUEJAYF4`. It stopped working due to 58ae4d494a15c591ada0cc71a0a3abed7fcbdecd. (I only get receiver telemetry (RSSI), but nothing from SmartPort (tension, current, altitude, GPS).) I did a `git bisect` between ~August 24 and current `master` to identify the change that broke it. Current `master` with 58ae4d494a15c591ada0cc71a0a3abed7fcbdecd reverted works for me; full telemetry restored.

jflyper, 2017-09-01:
I'll look into it.

jflyper, 2017-09-01:
@andrejpodzimek I have a fix #4036, but needs testing.

Here is a firmware hex for BLUEJAYF4, you may want to give it a try.
[betaflight_3.2.0_BLUEJAYF4-fix_softserial_ioconfiggpioaf.hex.zip](https://github.com/betaflight/betaflight/files/1269779/betaflight_3.2.0_BLUEJAYF4-fix_softserial_ioconfiggpioaf.hex.zip)

andrejpodzimek, 2017-09-01:
@jflyper  Yes, that works. I have telemetry with that build.

jflyper, 2017-09-02:
@andrejpodzimek Thanks for testing. I will merge the fix now.

closed 2017-09-02 by jflyper