betaflight/betaflight #4042: Beeper will not sound when activated via switch.

wydra91, 2017-09-02:
Okay, so I installed a two wire beeper on my Flip32 Omnibus F4 Pro v3, to the Buzz+ and Buzz- points, next to the boot button. I know the beeper works because when I power my FC I hear the usual tonal beep from the ESCs, and the three beeps that actually come from the beeper to indicate the gyro is calibrated.

I set a switch to a channel on my TX and it does register as AUX 3 on the Receiver tab.

In the mode tab, the Beeper mode is set to aux 3 and the row lights up when I flick the switch to the on position.

There is a blue light on the omnibus that starts blinking when I put it in beeper mode, but no sound comes out of the beeper.

I feel it's worth noting that the beeper does beep continuously when I turn of the transmitter, so the automatic lost function works, I would just like to be able to turn on the lost beeper with a switch as well.

Diff info below, I am fairly certain this is a bug as multiple people have chimed in on this problem on the /r/multicopter thread located [here](https://www.reddit.com/r/Multicopter/comments/6xhgh8/beeper_mode_wont_beep/). It seems to have started with 3.2

```
# diff

# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.2.0 Aug 28 2017 / 12:02:37 (b2cd7294e)

# name

# resources

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature RSSI_ADC

# beeper
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -MULTI_BEEPS
beeper -ARMED
beeper -SYSTEM_INIT

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 5 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 900 1050
aux 1 13 2 900 1075
aux 2 28 1 925 1150

# adjrange

# rxrange

# vtx

# rxfail

# master
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT600
set deadband = 3
set yaw_deadband = 2
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_vbat_pos = 2072
set osd_rssi_pos = 2050
set osd_tim_1_pos = 54
set osd_tim_2_pos = 33
set osd_flymode_pos = 2520
set osd_throttle_pos = 2434
set osd_vtx_channel_pos = 377
set osd_current_pos = 385
set osd_mah_drawn_pos = 2497
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2473
set osd_avg_cell_voltage_pos = 2066
set osd_pit_ang_pos = 2241
set osd_rol_ang_pos = 2123
set osd_battery_usage_pos = 2408
set osd_nheading_pos = 311
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_max_curr = OFF
set osd_stat_max_alt = ON

# profile
profile 0

# rateprofile
rateprofile 2

set rc_rate = 50
set rc_rate_yaw = 50
```

jflyper, 2017-09-02:
@wydra91 
Please RTFM about `beeper` command, `RX_SET` in particular.

And, if you actually hear 3 beeps for gyro calibration with this configuration, your configuration is dud, because you should not hear tones with `-ACC_CALIBRATION`. If you haven't disabled `RX_SET` nor `ACC_CALIBRATION` by yourself, I recommend you do CLI "defaults" and start configuring again.

For further assistance, please redirect your questions to the RCG thread. Thank you in advance.

Ref:
https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread

closed 2017-09-02 by jflyper

wydra91, 2017-09-02:
@jflyper While I appreciate your response clearing that up for me, the three beeps are for _gyro_ calibration in this case, _not accelerometer._ It wasn't until right before you posted that I discovered that RX_SET is the culprit. Might I recommend we clarify the tooltip for that one? It was the GPS keyword that caused me to turn it off, as I don't use GPS.

jflyper, 2017-09-02:
@wydra91 Glad you could make it work. The tooltip... description text ... may be confusing, but I have no trouble reading it (although the later half is some what hard to read for non-native like me). What do you recommend?

> RX_SET - Beeps when AUX channel is set, or beeps for number of satellite when GPS is enabled

wydra91, 2017-09-02:
If I knew how to program, I would look into making the feature more like airmode for example:

You can set airmode permanently on, but if you elect to leave that toggle off, you can enable it as a "mode" in the modes tab, in which case it is toggle able.

I would propose that the beeper function in the mode tab be separate from a GPS related beep, if possible. Unless I'm missing something (which I likely am)

The more I read the description the more that I kick myself for not catching it, unfortunately I'm not really sure if there is a better way to phrase it.

tantrim, 2017-10-04:
@jflyper  @wydra91 I too had this issue. The GPS in the tooltip threw me off as well for RX_SET. Although I am a developer myself this one got me. This is my first build so I have no knowledge of setting these things up yet so I'm looking at it with a fresh pair of eyes. This has thrown 2 people off that have taken the time to write about it.. I'm sure more might have this issue.

mikeller, 2017-10-05:
@wydra91, @tantrim: This confusion is addressed in #4219 (will also need a configurator change to get rid of the confusing tooltip).

wydra91, 2017-10-05:
@mikeller Wow, thanks you so much for taking the reigns on this one an knocking it out! I can't wait to see the change trickle through. You rock!

spits79, 2020-05-16:
Hello, I apologize for posting on such an old post and I am not sure if this is the right way to do it but I had exactly the same problem with BetaFlight 4.1.6 and Configurator 10.6.  The reason was there is two "RX_SET" setting is the Configurator one in the Beeper section and one in the Dshot Beacon section.  I had to turn both on before the AUX 3 channel would actually trigger a sound.  I actually use Oneshot ESC so I was ignored the Dshot section.  I just thought I would add my experience here.