betaflight/betaflight #4043: Omnibus F7 on 3.2 #1633 and 1623 only motor 1 spins

sdotfpv, 2017-09-02:
this is the dump i tried when i flashed. dont remember exactly which version i came from. but only motor 1 spins when armed. flashed back to rc4 and all 4 motors worked again. i noted that while on those 2 fws mentioned above escs would twitch when first plugged in but only escs on motor would have the start tone.
# Betaflight / OMNIBUSF7 (OBF7) 3.2.0 Aug  3 2017 / 09:20:41 (8b4ab25)

# name
name Sdot

# resources
resource BEEPER 1 D15
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 E09
resource MOTOR 4 E11
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 D12
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 NONE
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
resource I2C_SDA 1 NONE
resource I2C_SDA 2 NONE
resource I2C_SDA 3 NONE
resource I2C_SDA 4 NONE
resource LED 1 E00
resource LED 2 NONE
resource LED 3 NONE
resource RX_BIND 1 NONE
[... 550 lines cut ...]
set horizon_tilt_expert_mode = OFF
set gps_pos_p = 15
set gps_pos_i = 0
set gps_pos_d = 0
set gps_posr_p = 34
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83

# rateprofile
rateprofile 1

set rc_rate = 110
set rc_rate_yaw = 100
set rc_expo = 30
set rc_expo_yaw = 30
set thr_mid = 40
set thr_expo = 0
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 77
set tpa_rate = 10
set tpa_breakpoint = 1650

jflyper, 2017-09-02:
@sdotfpv 

I think you are really confused.

> Betaflight / OMNIBUSF7 (OBF7) 3.2.0 Aug 3 2017 / 09:20:41 (8b4ab25)

This firmware is way back, about a month old.

closed 2017-09-02 by jflyper

sdotfpv, 2017-09-04:
@jflyper  i tried with this dump and from scratch and only motor 1 was spinning.. on firmware #1633 i only pasted the dump so u can see what i did.. i am deff not confused bud

jflyper, 2017-09-04:
Can you reflash again with the problematic firmware or do cli "defaults", and immediately test the motors without configuring anything else? Start with OS125.

sdotfpv, 2017-09-04:
yes thats what i did the second time after using this initial cli dump. full chip erase is always selected. manually reconfigured settings and still only motor 1 spun.  after this i  just flashed RC4 and everything worked again.

jlbiarlee, 2017-09-04:
I am having similar problems- how many Uarts are you using -

sdotfpv, 2017-09-16:
@jlbiarlee  just tried RC5 same thing happend switched to rc4 and everything fine