betaflight/betaflight #4044: BF 3.2 RC4 LED color controlled from AUX channel issue

Jahnkeanater, 2017-09-02:
BF 3.2 RC4 LED color controlled from AUX channel. In BF 3.1.7 an aux channel had the ability to cycle through every possible color. It seems like in BF 3.2 that you can only cycle through 2 or 3 colors. It also seems like the default color is when the aux channel is centered rather than default at 1000.

This kind of ruins the aux feature since there is really no point in having several shades of the exact same color rather than being able to actually change the color.

I have the LEDs setup on aux 4 and sweeping from 1000 to 2000 slowly fades from white to red.

Jahnkeanater, 2017-09-02:
It also seems like the VTX to assign color feature does nothing using smart audio.

MrPlocki, 2017-09-02:
I am also experiencing this issue with assigning colour to LEDs via an aux channel. When cycling through colours, it appears that the value for one of the three colour channels is being lost somewhere. This seems to be the case due to the LEDs being unlit at a point where they should be showing a solid colour.

mikeller, 2017-09-03:
Please read https://github.com/betaflight/betaflight/wiki/LED-Strip-functionality

MrPlocki, 2017-09-04:
Thanks for the reply. I like the new functionality to fade between defined
colours.
I am now able to get a fade between full rainbow hue spectrum using aux
channel.

There does seem to be a problem when using aux modifier with arm state
function. The colour will fade to whatever is set as colour 0 in the
palette. This happens for both armed and disarmed states. The fade occurs
in both directions.

i.e. 1500 = disarmed colour
      1000 = palette colour 0
      2000 = palette colour 0

This happens whatever palette colour you choose.

I think this was causing me to not understand the new functionality as I
always use arm state function LEDs.

Thank you for the great work,

Matt

tionality

mikeller, 2017-09-04:
@MrPlocki: I think this is to be expected, since the fade functionality is now mixed into the colour that results after all other colour modifiers are applied. This means that not all possible combination of modifiers 'make sense', or give good looking results.

MrPlocki, 2017-09-04:
Sorry, I really like arm state LEDs, so I had a look at ledstrip.c

If "case LED_FUNCTION_COLOR" is not run then values for "nextColor" and "previousColor" are not generated. This is an issue later executing   "ledGetOverlayBit(ledConfig, LED_OVERLAY_THROTTLE)" which uses these variables. 
If you agree, could you please add the "nextColor" and "previousColor" variables to other cases or the led overlay "if" statement. 

Did I mention I like arm state LEDs?

mikeller, 2017-09-05:
Well spotted, @MrPlocki, that is a bug indeed!

Would you be able to submit a pull request for a fix?

MrPlocki, 2017-09-05:
@mikeller I'll have a go, might be a bit beyond me. The best I can come up with is to assign arbitrary palette colors to the arm function. i.e. 
`nextColor = ledStripConfig()->colors[0];`

This would not be the neatest solution, but the functionality would improve for arm state leds. I would also assign different pairs of arbitrary colors for armed and disarmed states, allowing for full led customization. 

Not sure how to get the `LED_SCOLOR_ARMED` etc. as an integer to increment, or i would do it that way. This would be more consistent functionality.

I'm a bit new to this, any suggestions welcome.

Rakinroul, 2017-10-08:
Hello guys.
I apreciate your work but can you set it like it was before ?
I mean thru aux when set on color to be able to go thru whole pallete , when other values will be same ?

Thanks

mikeller, 2017-10-08:
@Rakinroul: Yes you can. When choosing a fixed colour with a throttle / aux fader overlay, just choose the colour for the LED to be the middle colour of the fading range, and the colours to the left / right of it to be the start point / end point of the fading range. 

See also: https://github.com/betaflight/betaflight/wiki/LED-Strip-functionality

closed 2017-10-08 by mikeller