betaflight/betaflight #4047: [Question] Gyro filter latency: Software vs. Hardware

robert-hoffmann, 2017-09-02:
Hi,

I just managed to remove all notch filters, and switch all remaining filters to PT1 without any heating issues.

So i was wondering, how much more advantage (latency-wise) would you get in deactivating the gyro soft filter, and replacing it with the gyro hard filter ?

Anyone have an idea ?
Thanks !

Arakon, 2017-09-02:
I believe the reason we moved from using the hardware filters to software filters was that the software filters have less delay, and introduce less aliasing in the data.

Hardware filters were originally used, long ago.

ledvinap, 2017-09-02:
Delay caused by filter with given characteristic is about the same independent of filter realization (software/mechanical). Notch filter may be better that low-pass, it has lower group delay at high frequencies and maybe a bit les delay at notch (I'm not sure ..)

Downsampling data will cause aliasing of frequencies over Nyquist, so there frequencies shall be removed before discarding samples. With low gyro sampling (over SPI/I2C) mechanical filtering or internal filter help considerably.

robert-hoffmann, 2017-09-05:
Thanks for the feedback, i guess i'll leave things as the currently are

closed 2017-09-05 by robert-hoffmann