betaflight/betaflight #4048: Dynamic filter + dshot150 causing motors to stop working?

silenec, 2017-09-02:
Hi, I am not sure if this is a configuration, hardware or software fault but here is my weird problem:
BF3.2.0-RC4, Omnibus F3, Racerstar RS30A V1, 2205 motors, 4S battery.
I flew this setup for a year with older versions of BF but now with 3.2.0RC4:

Oneshot, multishot, dshot150 works ok on 4/4/4 and 8/4/4kHz with almost any configuration (except the ones below):

- dshot150 4/4/4 with dynamic filter enabled = motor number 1 in BF stops working after a few seconds.
- dshot150 4/4/4 with dynamic filter enabled and accelerometer disabled (lower cpu load) = motor number 1 in BF stops working after a few seconds AND motor number 2 in BF also stops working soon after that

Also:
- dshot150 4/4/4 with dynamic filter disabled and acc on or off = no problems at all
- dshot150 8/4/4 with dynamic filter enabled and acc on or off = no problems at all
- oneshot42 or multishot 4/4/4 and 8/4/4 with dynamic filter enabled and acc on or off = no problems at all

I tried it both without and with the signal grounds connected, no change.
- once the motor stops working, I have to disconnect BOTH the battery and USB power and then they start working again immediately
- if I only disconnect the battery and keep USB power connected, the motors do not start working
- changing the ESC protocol from d150 to any other protocol makes the motor start working again  even without disconnecting the power

Anyone here with any idea of what the problem might be?

Thanks

\\\
# diff
# Betaflight / OMNIBUS (OMNI) 3.2.0 Aug  6 2017 / 13:33:02 (981df1d54)

feature -RX_PPM
feature RX_SERIAL
feature MOTOR_STOP
feature ANTI_GRAVITY
feature DYNAMIC_FILTER
beeper -ON_USB
serial 2 64 115200 57600 0 115200
aux 0 0 0 1050 1250
aux 1 28 0 1125 1500
aux 2 2 0 900 1125
aux 3 13 1 1150 1250
aux 4 26 0 1050 1250
set gyro_sync_denom = 2
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set rssi_channel = 9
set max_aux_channels = 5
set serialrx_provider = SBUS
set sbus_inversion = OFF
set min_throttle = 1085
set motor_pwm_protocol = DSHOT150
set motor_pwm_rate = 8000
set failsafe_throttle = 1200
set failsafe_throttle_low_delay = 200
set align_board_yaw = 90
set bat_capacity = 1500
set current_meter = VIRTUAL
set ibatv_scale = 400
set ibatv_offset = 100
set small_angle = 180
set auto_disarm_delay = 30
set disarm_kill_switch = OFF
set deadband = 2
set pid_process_denom = 1

profile 2

set dterm_lowpass_type = PT1
set yaw_lowpass = 70
set p_pitch = 70
set d_pitch = 22
set p_roll = 54
set d_roll = 20
rateprofile 0

set rc_rate_yaw = 110
set thr_mid = 0
set thr_expo = 50
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80

# 
\\\

silenec, 2017-09-02:
Video: https://www.youtube.com/watch?v=gzsjWj7ver0

diggit, 2017-09-06:
Hi,
I noticed similar behavior on motor 1. After arming, motors spins up and works for several seconds, then motor 1 stops (up to ~15s so I managed to take off :/).
After re-arming, same scenario repeated. It was happening till I disconnected and reconnected battery (power cycle). So it is not so easy to replicate.

I noticed, that motor1 esc (DYS XSD 20A) stopped and beeped like after boot (data stream recovered?). 

Using OMNIBUSF4SD 3.2.0.RC? (same build date as OP), DSHOT600, 8/8kHz and dynamic filter enabled.

silenec, 2017-09-06:
Yeah I managed to take off twice before I figured out what was happening. Luckily everything survived. https://youtu.be/tGOsOMugBCc
My motor did not beep at all and I could not get it to run again without power cycling the FC.
I am now running multishot just to be safe and flew 10 packs without any problem. Dynamic filter with DSHOT150 simply does not work for me. DSHOT150 without the dynamic filter worked fine since the day that dshot came out though.

diggit, 2017-09-06:
Mine motor beeped right after it stopped, so I noticed it only when testing on ground. I can hear some beeping when your drone hits ground. What is that? (not heard in first video, probably sth not important)

silenec, 2017-09-06:
The beeping in the video before it hits the ground is me disarming the quad.

flightingflight, 2017-09-22:
Hi, it looks like I have a similar issue (motors stopping) except that for me, if I run the same gyro and pid loop at the same rate say 4k/4k or 2k/2k, it can cause 1 or 2 motors to stop spinning after a few seconds (props off), it doesn't matter if dynamic filter is on or off. I also tried to raise my idle to 7, no change. Flashed Betaflight 3.2 RC5 

Once I change to 8K/4K or 4K/2K the issue goes away (and can leave idle back to 4.5). I can replicate the issue every time I have tried (at least 10 times). The way I am testing, is just having the quad powered off a 12v power supply only and I have to power cycle the quad for the motors to spin again.

I have been using 3.1.7 for a few months on 8K/4K with Dshot150 (signal capacitors left on) and 4.5 idle without any issues.

Omnibus F3 AIO v1.1
Emax RS2205 2300KV Red bottoms
Turnigy MultiStar 20A BLHeli_S (G-H-30) running Dshot150

```
# diff
# Betaflight / OMNIBUS (OMNI) 3.2.0 Sep 11 2017 / 06:43:51 (649ccd7c7) MSP API: 1.36

feature -RX_PPM
feature RX_SERIAL
feature ANTI_GRAVITY
beeper -ON_USB
serial 2 64 115200 57600 0 115200
aux 0 0 0 1800 2100
aux 1 28 1 1800 2100
aux 2 4 1 1800 2100
aux 3 13 1 900 1200
set gyro_sync_denom = 2
set acc_hardware = NONE
set acc_trim_pitch = 2
set baro_hardware = NONE
set min_check = 1015
set rssi_channel = 7
set rc_interp_ch = RPYT
set serialrx_provider = IBUS
set dshot_idle_value = 700
set motor_pwm_protocol = DSHOT150
set vbat_max_cell_voltage = 45
set vbat_min_cell_voltage = 30
set vbat_warning_cell_voltage = 30
set small_angle = 180
set deadband = 4
set yaw_deadband = 6
set pid_process_denom = 1
profile 0

set dterm_lowpass_type = PT1
set dterm_setpoint_weight = 100
rateprofile 0

set rc_expo = 25
set rc_expo_yaw = 25
set thr_mid = 0
set thr_expo = 30
set roll_srate = 82
set pitch_srate = 82
set yaw_srate = 55
```

flightingflight, 2017-09-24:
Solved my problem, I recall people having issues with blackbox logging to the SD card in regards to dma and what not. I turned off blackbox logging completely and the issue is gone. But it would be nice to have blackbox logging working.

I was logging at 1KHz before, tried logging with 500Hz, with sdcard_dma on and off, no change.

fiveangle, 2017-09-24:
I had similar issues with 1 or 2 motors disabling shortly after arm, resuming normal operation only after power-cycle of FC.  OMNIBUS F3.  I didn't put much thought into it since it was running on default loop times and after switching to my standard (8/4/4 DSHOT600) problem went away.  If the problem surfaces again I will take the time to investigate based on this info that it may be related to dshot-looptime-filter interaction.

silenec, 2017-09-24:
I use 16.6

fiveangle, 2017-09-26:
?  Are you sure ?  If so, upgrade to 16.63 or 16.67 and see if same results ?  I'm thinking I may have encountered this issue when on 16.63, but can't remember now (I'm on 16.67 now).  I haven't seen it in a few weeks and can't seem to repro now so wondering if it's related to BLheli interaction (perhaps motor 1 / 2 are being sent different info when the bug occurs, but doesn't affect 16.67 and does 16.63 ?  Grasping at straws, I know...

mikeller, 2017-09-26:
@fiveangle: There were a number of different versions released as '16.63'. The earlier ones had bugs, hence they got replaced (unfortunately without making it easy for users to see by giving them new numbers). If you download the latest 16.63 now it should work properly.

silenec, 2017-10-14:
Ok here is an update. I have replaced my PDB with a new one, removed the input capacitors on ESCs and flashed them to 16.63 FW. Also flashed the latest Betaflight 3.2.0 and everything seems to be working fine now :) I think it might be the capacitors so if there's anyone running dshot with capacitors still in place, I recommend that you remove them even if it works for you without problems - I ran D150 for a year and once the dynamic filter came out, I started having problems. Some combinations of settings worked and some did not, so it is better to be safe and remove the capacitors now (if you are running dshot).

mikeller, 2017-10-14:
@silenec: Can this issue be closed?

silenec, 2017-10-14:
Since this turned out to most likely not be a software bug, I think we can close this.

closed 2017-10-14 by mikeller

diggit, 2017-10-15:
Today, I flashed OMNIBUSF4SD with stable release 3.2.0 and loaded backed up config using cli.

Highlights:
DSHOT 600
gyro 8kHz
PID 8kHz
logging 8kHz

 
```
# diff

# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.2.0 Sep 28 2017 / 16:38:50 (ceb6f70) MSP API: 1.36

# name
name ORION

# resources
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource LED_STRIP 1 A01
resource INVERTER 3 NONE

# mixer

# servo

# servo mix

# feature
feature -OSD
feature TELEMETRY
feature LED_STRIP
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 2 32 115200 57600 0 115200

# led
led 0 5,5::A:2
led 1 5,6::A:2
led 2 5,7::A:2
led 3 5,8::AB:2
led 4 5,9::CB:6
led 5 10,5::A:2
led 6 10,6::A:2
led 7 10,7::A:2
led 8 10,8::AB:2
led 9 10,9::CB:6

# color
color 2 1,0,255

# mode_color
mode_color 6 0 0
mode_color 6 1 6

# aux
aux 0 0 0 900 1100
aux 1 1 1 1900 2100
aux 2 26 1 1400 1600

# adjrange
adjrange 0 0 2 900 1100 6 3
adjrange 1 0 2 1400 1600 7 3
adjrange 2 0 2 1900 2100 8 3

# rxrange

# vtx

# rxfail

# master
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 120
set gyro_notch2_hz = 0
set serialrx_provider = SBUS
set dshot_idle_value = 350
set motor_pwm_protocol = DSHOT600
set align_board_yaw = -90
set small_angle = 180
set pid_process_denom = 1
set debug_mode = NOTCH

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_notch_hz = 0
set dterm_setpoint_weight = 60

# rateprofile
rateprofile 0

set rc_rate = 120
set rc_rate_yaw = 120
```

I depleted 2 batteries without problem. During 3rd one, everything went ok, but suddenly MOTOR 1 stopped and quad started spinning in air. (and hit the ground)
Weird thing is, once this problem happens, and you just rearm, there is quite high chance, it will happen again. In my case like 6 time in row in range of ~10 to ~60 seconds after arming.

Once the quad hit ground, (most probably) ESC 1 beeped like at power up.

After several attempts of taking off, battery was empty.

With next battery, everything worked well.

2 logs attached, interesting is the end where motor 1 hits max. That makes sense, PID tried to compensate for low motor thrust, but there was no response. But what can cause this?

Changed .BFL to .TXT for direct upload here.
[LOG00026.TXT](https://github.com/betaflight/betaflight/files/1385509/LOG00026.TXT)
[LOG00027.TXT](https://github.com/betaflight/betaflight/files/1385510/LOG00027.TXT)

Should I open new issue?