betaflight/betaflight #4049: LED strips are not working as expected with SPRACINGF3EVO 3.1.7 on Eachine v-tail 210

0zoon0, 2017-09-02:
Hi guys,
I've flashed the betaflight SPRACINGF3EVO 3.1.7 10-04-2017 13:36 on an Eachine v-tail 210 and the LED strips are not working as expected: four of the LEDs are lighting white (and sometimes the fifth led is yellow, but most of the times is off). The quad has three LEDs on each front arms facing down, and four on the tail (facing south). All three LEDs on the right arm are always white (which are 0-2 leds), and one LED on the left arm is white (I suppose that one is index 3, just guessing), and sometimes (can't tell under what conditions) one is yellow (but most of the time is off), and the third one is always off. The tails LEDs are always off.

The diff is:
\\\
# diff
# Betaflight / SPRACINGF3EVO 3.1.7 Apr  3 2017 / 22:28:08 (e1c4b5c)
name vtail210

feature -TELEMETRY
feature -CURRENT_METER
feature GPS
feature LED_STRIP
serial 0 1 115200 57600 0 115200
serial 2 2 115200 57600 0 115200
led 0 15,6:ED:AI:6
led 1 15,7:ED:AI:6
led 2 14,7:ED:AI:6
led 3 0,6:WD:AI:6
led 4 0,7:WD:AI:6
led 5 1,7:WD:AI:6
led 6 6,7:S:AT:12
led 7 7,7:S:CT:12
led 8 8,7:S:CT:12
led 9 9,7:S:AT:12
aux 0 0 0 1775 2100
aux 1 2 1 900 1300
aux 2 13 2 1700 2100
rxfail 6 s 2000
set min_throttle = 1050
set motor_pwm_protocol = ONESHOT125
set motor_pwm_rate = 2000
set disarm_kill_switch = OFF
set gps_provider = UBLOX
set failsafe_throttle = 1300
set failsafe_procedure = AUTO-LAND
set acc_trim_roll = 14
set baro_hardware = AUTO
set mag_hardware = AUTO
set magzero_x = 135
set magzero_y = 81
set magzero_z = -156
profile 0

rateprofile 0
rateprofile 0
\\\

Note that the same behavior (four LEDs are white, all other off, sometime the fifth could be yellow) is there regardless of LED strip settings in betaflight (I tried different setups).
Just before flashing betaflight, it was running cleanflight SPRACING3EVO 1.13.0 June 6 2016 (configurator version 1.2.4) and all LEDs were working as expected.

Thank you.

jflyper, 2017-09-03:
@0zoon0 Please redirect your issue report and ask for help at some other appropriate places on the net, such as a RCG thread, or contact your distributor for the following reasons.

1. This is not a place for user support, but for software improvement.

2. Your FC is probably a clone or design-clone of the authentic SP Racing F3 EVO, and we don't know what has been changed from the original design.

3. Your FC came flashed with a version of CF.

Thank you in advance.

Ref:
https://www.rcgroups.com/forums/showthread.php?2767388-Eachine-V-tail-210-FPV-Drone-SP-Racing-F3-DVR-What-Amazing-Unique-Design%21%21%21

EDIT
Oh, you seem to have a duplicate question there.

closed 2017-09-03 by jflyper

0zoon0, 2017-09-03:
Hi jflyper,
Thank you for quick reply.
Indeed, I asked on forums first and nobody could give advise, so I opened this issue.
Regards.

jflyper, 2017-09-03:
@0zoon0 I'm following you up in the thread.