betaflight/betaflight #4051: Kakute F4 AIO 3.2RC4 Reduced power

michael-blue2, 2017-09-03:
After installing the 3.2RC4 on my kakute f4 aio it seems i have way more oscillations and way less power on full throttle. I tried reducing all the pids but still i get a lot of vibrations and way less power. As with 3.1.7 and 3.1.6 i couldnt get my smartport telemetry working with softserial and remapping. The telemetry works only on 3.2RC4

TimTimNL, 2017-09-17:
@michael-blue2 I can't get any telemetry working with the board and feel like i've tried everything flashing my RX,TX and CLI commands

michael-blue2, 2017-09-17:
@TimTimNl 
copy paste this in the cli with betaflight 3.2. Solder the s.port cable to the led output
CLI:
feature SOFTSERIAL
resource LED_STRIP 1 NONE
resource SERIAL_TX 11 A01
serial 30 32 115200 57600 0 115200
set tlm_inverted = OFF
set tlm_halfduplex = ON
save
I will have to close this thread as its wrong and my bearing were just junk.
If you have any questions use the rc group thread.
https://www.rcgroups.com/forums/showthread.php?2904475-Holybro-Kakute-F4-AIO-Flight-Controller/page24

closed 2017-09-17 by michael-blue2