betaflight/betaflight #4057: BF 3.2 RC4 - BiBlade setups are unfliable and untunable

kidBrazil, 2017-09-03:
I have a fleet of quads with the same components. Yesterday I built a fresh one and flashed RC4. I run 5050C Dal Biblades exclusively on this fleet but when I went to fly/tune the quad was unfliable and untunable. Whenever throttle went over 50% it would just start shaking like crazy. Here is a link to BB logs ( https://www.dropbox.com/sh/3z9kws45sjfjlav/AABYuTYLpGhtrYribm5LNVMwa?dl=0 ). After checking all manner of hardware I switched props to Dal Q5040 and the problem went away. I was able to tune and fly.

This afternoon I re-flashed the same quad to RC2 and put the BiBlades back on. It worked flawlessly just like the rest of my fleet that is also on RC3/RC2.

I built a DMR Doberman 5" stretch and flashed RC4 - Using the same BiBlades quad was unfliable again. Flashed RC2 and problem went away.

## Components
Mia-X Frame
T2 2206/2600
DAL 5050C BiBlade
Cicada 35A Bl_S ESC
CL_Racing F4 FC
R12DSM RX ( SBUS)
HS1177 + VTX03 

## Reproduction
Grab a quad, flash RC4, basic config... add 5050C Biblade or some other biblade and see if it flies...

## Diff Config (works)
# diff

# version
# Betaflight / CLRACINGF4 (CLR4) 3.2.0 Aug  1 2017 / 09:34:39 (869dd1f)

# name

# resources

# mixer

# servo

# servo mix

# feature
feature -OSD
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -MULTI_BEEPS
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE

# map

# serial
serial 0 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 2 1700 2100

# adjrange

# rxrange

# vtx

# rxfail
rxfail 6 s 1000

# master
set gyro_notch1_hz = 0
set acc_hardware = NONE
set min_check = 1020
set max_check = 2000
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT600
set deadband = 10
set yaw_deadband = 10
set pid_process_denom = 1

# profile
profile 0

set dterm_lowpass_type = PT1
set anti_gravity_gain = 2800
set i_pitch = 60
set d_pitch = 28
set p_roll = 58
set i_roll = 60
set d_roll = 28
set i_yaw = 60

# rateprofile
rateprofile 0

set rc_rate = 143
set rc_rate_yaw = 143
set rc_expo = 20
set rc_expo_yaw = 20
set roll_srate = 77
set pitch_srate = 77
set yaw_srate = 77

jflyper, 2017-09-03:
- Do motors spin up smoothly beyond 50% throttle without props?

- Do motors spin up smoothly beyond 50% throttle with props?

kidBrazil, 2017-09-04:
Yes with slider motors spin no worries both with or without props.

I don't have a reliable way to check balance. But I used fresh 5050C Dal Biblades from a different batch, also clear instead of black.

Only issue arises from quad being armed and flying on RC4. RC2/RC3 no problem. I have some friends who are going to try and replicate as well

kidBrazil, 2017-09-04:
@jflyper I also flashed RC4 On a few other of my fleet and they all had issues with BiBlades, absolutely unfliable. - Good to go again once I switched back to RC2

martinbudden, 2017-09-04:
@kidBrazil , when you say untunable, did you try setting all the filters to default before you started to tune?

kidBrazil, 2017-09-04:
@martinbudden Yeah I did, the maiden was on a fresh RC4 flash with nothing changed except for the most basic of config so I could arm it and using my rates.

Biquad, PT1, Dynamic Notch... TPA.. PID's no matter what I tried to change after that it still didn't make it flyable. Basically, as soon as I reached 50% throttle on biblades, it would just start shaking so much, and had no power at full throttle.

I took the same quad and put DAL Q5040's on it and suddenly its fine. Flies beautifully with ZERO issues or warbles.

Grabbed same quad again, switched back to biblades - issues all over again. Then I flashed RC2 on it and it had no issues with the Biblades or the quadblades. It flew as I have come to expect from these builds. I have been using the same components for months now.

martinbudden, 2017-09-04:
Hmm. I'm not sure what could be the cause, since not much has changed to do with tuning between RC2 and RC4. Just to eliminate it as a possibility, can you try `set iterm_windup = 250`? I don't really think this will solve it, but it is one of the things that changed so it is worth eliminating.

kidBrazil, 2017-09-04:
@martinbudden - I will take a look at that for sure so we can start eliminating variables. Might be tomorrow tho =S

I've been asking other people to test and same reports.
[image]

kidBrazil, 2017-09-06:
@martinbudden @jflyper  I am getting more reports from people of the same sort of behavior, check out the video I posted here and check the comments - https://youtu.be/NJC00qlF-zY

So this is definitely not isolated to just my case and I have been able to reproduce this issue on all of the quads that I flash RC4 on. Motor soft mounting is an option but.. up until now quite unecessary with Betaflight.

dcanham, 2017-09-06:
Hey everyone, canham_fpv here. I never got BB to work on RC4, but when I flashed back to RC3 it worked just fine. On triblades, I had a random yaw twitch and on biblades it was unflyable past 50%. I couldn't get any BB from it and the DVR didn't autostart in time to catch anything good for that. But now that I'm on RC3 I haven't had a yaw twitch (yet) and BB works. Will try biblades tomorrow during practice and when it finally quits raining...

brycedjohnson, 2017-09-07:
If you wanted to try a somewhat annoying binary search, you could load some builds between rc3 and rc4.

RC3 was Aug 6 and RC4 was Aug 27th

Maybe try this one on aug 16th to start with?  

https://ci.betaflight.tech/job/Betaflight/83/

Make sure you do a full erase between each time.

mikeller, 2017-09-07:
@brycedjohnson: `git bisect` will make this a lot less annoying.

martinbudden, 2017-09-07:
@kidBrazil , from report it is clear that the problem exists with RC4 and does not exist with RC2. However it's not 100% clear what the status for RC3 is. Can you confirm that RC3 works (or not)?

kidBrazil, 2017-09-07:
@martinbudden I have reports on the BF slack that RC3 works. I can do my own testing hopefully tomorrow if the weather cooperates.

toiminto, 2017-09-07:
https://www.youtube.com/watch?v=l-ZOyBfdFPM Here is some footage with similar issues I had on cl racing f4 flashed with bf 3.2 rc4 - commented the video with timestamps, basically after punch out and dropping throttle did some weird uncommanded moves. I thought it was anti gravity at first and turned it off but it still did similar uncommanded moves/twitching after that

martinbudden, 2017-09-07:
@kidBrazil , @toiminto , we need some blackbox logs of the problem so we can try and fix it.

toiminto, 2017-09-07:
@martinbudden I will try to reflash rc4 and do logs/reproduce problem next week. Will post any findings here

martinbudden, 2017-09-07:
@kidBrazil , @toiminto , when you test, can you use the latest build rather than RC4, see: https://ci.betaflight.tech/job/Betaflight/lastSuccessfulBuild/artifact/obj/

kidBrazil, 2017-09-08:
Hey Guys, I just got back from testing the latest version that @martinbudden linked. Results were the same, completely unfliable with twoblades. It has no power, any time I try to go full throttle it just warbles and has maybe 50% of the power.

I did a stock PID run with a fresh flash of RC4 ( latest ) and configured it using the nightly configurator. Still same results with RC4. RC3 flew fine, I even limped back with a prop that had lost a good .5" of the tip on one of the two blades. In fact, flying RC3 with a half broken prop is what it feels like to fly RC4 with brand new twoblades.

So I got a BB log for you guys that I will link here -https://www.dropbox.com/s/ah5hza5rxh3l7wd/LOG00001.BFL?dl=0

And I got a video of the flight Here (SOUND WARNING) - https://youtu.be/DdxoopiyCYM

I was able to make this version of RC4 semi fliable ( at least no warbles ) by using the following PID - P20 I10 D0 - on all axis.

I did not try PT1 or messing with notch filters as I had no time.. but this is strange. Same quad on RC3 or RC2 flies perfectly fine with stock settings and Biquad.

# Betaflight Dump
# version
# Betaflight / CLRACINGF4 (CLR4) 3.2.0 Sep  8 2017 / 13:20:59 (759788fe2)

# name
name -

# resources
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 B08
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 B08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource INVERTER 1 C00
[... 1282 lines cut ...]
set horizon_tilt_expert_mode = OFF
set gps_pos_p = 15
set gps_pos_i = 0
set gps_pos_d = 0
set gps_posr_p = 34
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83

# rateprofile
rateprofile 0

set rc_rate = 143
set rc_rate_yaw = 143
set rc_expo = 20
set rc_expo_yaw = 20
set thr_mid = 50
set thr_expo = 0
set roll_srate = 77
set pitch_srate = 77
set yaw_srate = 77
set tpa_rate = 10
set tpa_breakpoint = 1650

tigert, 2017-09-08:
Just for the record, I have a RC4 on a CL Racing F4 here and it works fine. Gemfan Flash, bith tri and dualblades, when tested. Dyn filters and PT1.

martinbudden, 2017-09-09:
@kidBrazil , can you post the output from the CLI `tasks` command here?

Also, next time you test, can you try using multishot, to see if it makes a difference?

kidBrazil, 2017-09-09:
@martinbudden I will try again on monday with your suggestions. Don't think I will have time before that unfortunately

martinbudden, 2017-09-09:
@kidBrazil , great. Also, I forgot to mention, could you also try 4kHz looptime as well?

kidBrazil, 2017-09-10:
@martinbudden Hey man, so I took the drone out again today after re-flashing it with latest RC4 and using the latest configurator and STILL the exact same issues.

4khz loop times did not help, Multishot did not help either. I am sort of at a loss here especially since RC3 or RC2 work perfectly and fly wonderfully.

As requested, here is the "Tasks"

# tasks
Task list             rate/hz  max/us  avg/us maxload avgload     total/ms
00 - (         SYSTEM)      9       4       0    0.5%    0.0%         0
01 - (            PID)   3937      44      30   17.8%   12.3%      7266
   - (           GYRO)   7874
04 - (             RX)     49      34      27    0.6%    0.6%        52
05 - (         SERIAL)     99      94       1    1.4%    0.5%         6
07 - (BATTERY_VOLTAGE)     49       4       2    0.5%    0.5%         2
08 - (BATTERY_CURRENT)     49       3       1    0.5%    0.5%         1
09 - ( BATTERY_ALERTS)      4       4       2    0.5%    0.5%         0
10 - (         BEEPER)     99       4       1    0.5%    0.5%         2
17 - (            CMS)     59       4       1    0.5%    0.5%         1
18 - (        VTXCTRL)      4       3       0    0.5%    0.0%         0
19 - (        CAMCTRL)      4       1       1    0.5%    0.5%         0
RX Check Function                   6       1                         2
Total (excluding SERIAL)                        22.4%   15.9%

flint723, 2017-09-12:
Did you get a chance to try RC5 yet?

kidBrazil, 2017-09-12:
@flint723 @martinbudden Not yet, I will give it a try this week and hopefully it sorts out the issue.

kidBrazil, 2017-09-13:
Ok guys, I got out today and test RC5 and I am happy to report that altough the stock tune wasn't great on Biblades it was flyable and I was able to reach full throttle. There were some warbles for sure at higher throttle ranges but I was able to tune it out quite easily.

So just to recap here real quick.
1- This is still that same quad, with the same motors and same issues on RC4 that flew great on RC2/RC3
2- I flashed RC5 while using the latest configurator available on the master branch
3- I used the same configurator to make all the settings and tuning adjustments.

## Tuning the quad
Tuning wasn't all that difficult, these are the steps I took to get my quad to stop warbling and behave quite well. Minimal propwash and I tuned it in low to moderate wind.

1- Enabled Dynamic Filters
2- Enabled PT1
3- DIsabled Notch 1 & Notch 2 
3- Drop D term to ~15 ( less is more here..)

Once there I started raising P gains until I got fast oscillations, backing it up a bit and then I barely even touched D gain. ended up with what you see below on the diff.

### Quad is flying great on the 5050C Biblades. Here is a video post tune - https://youtu.be/fOKiX8TLt5k

### Here's a diff on the quad

# diff

# version
# Betaflight / CLRACINGF4 (CLR4) 3.2.0 Sep 11 2017 / 06:31:34 (649ccd7c7) MSP API: 1.36

# name

# resources

# mixer

# servo

# servo mix

# feature
feature -OSD
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -MULTI_BEEPS
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE

# map

# serial
serial 0 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 2 1700 2100

# adjrange

# rxrange

# vtx

# rxfail
rxfail 6 s 1000

# master
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 0
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 0
set acc_hardware = NONE
set min_check = 1020
set max_check = 2000
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT600
set align_board_yaw = 180
set deadband = 10
set yaw_deadband = 10
set pid_process_denom = 1

# profile
profile 0

set dterm_lowpass_type = PT1
set anti_gravity_gain = 3200
set p_pitch = 50
set i_pitch = 60
set d_pitch = 20
set p_roll = 50
set i_roll = 60
set d_roll = 20

# rateprofile
rateprofile 0

set rc_rate = 143
set rc_rate_yaw = 143
set rc_expo = 20
set rc_expo_yaw = 20
set roll_srate = 77
set pitch_srate = 77
set yaw_srate = 77

ctzsnooze, 2017-09-14:
@kidBrazil I looked at your logs, from RC4.  It shows massive sinusoidal oscillation at 30Hz.  It is primarily generated from D.

The main thing is that the oscillation is exclusively in the pitch axis.  There is none in roll or yaw.  This is very unusual.  It implies that the flight controller / gyro chip is somehow loose or in a low pass state on this axis, and that when props are applied that maybe are a bit out of balance, the system can resonate at a very low frequency.

I am confident that on your 'normal' props that fly OK, we will see the same phenomenon.  

I anticipate this being a hardware issue with the quad that is tolerable with tri-blades but made more obvious with bi-blades.  

Playing with filtering won't fix it since the frequency is very low and the amplitude massive.  There is no evidence of any other abnormal behaviour in the log.

kidBrazil, 2017-09-14:
@ctzsnooze Thank you very much for the indepth explanation. I will look at the FC once again and maybe replace it with something else!

Cheers

fftunes, 2017-09-15:
@ctzsnooze @kidBrazil But that doesn't explain how the issue could be reproduced on other quads.

kidBrazil, 2017-09-16:
@fftunes agreed, however I am getting reports that flashing to RC5 works... I dunno why either.

fftunes, 2017-09-17:
Any info on what was changed from RC4 to RC5? Respectively what kind of bug went into RC4?

From the symptoms described here, and other posts on rcgroups, it sounds a bit like yaw P got scaled by accident?

martinbudden, 2017-09-20:
Closing, since reports are that problem is fixed on RC5.

My view is that the problem was caused by the jitter that resulted from the fact that `DEBUG_GYRO_NOTCH` was set on in RC4.

closed 2017-09-20 by martinbudden