betaflight/betaflight #4059: MINI BRF4(BeeRotor F4) cannot use TBS SmartAudio ?

DooDoo512, 2017-09-04:
mini BRF4无法使用TBS SmartAudio功能。

1.) My board is mini BeeRotor F4; VTX is support TBS SmartAudio to change frequency band or channel.
我用的是miniBRF4飞控；图传支持 TBS SmartAudio 调发射频率。

2.) My XM+ reciver is UART rx2 with SBUS, VTX connecting to tx1  which TBS SmartAudio selected, when use my Frisky X9D+ select VTX SA on the OSD menu. In the submenu VTX SMARTAUDIO this is noting except "- -- ---- ---" and STAT & BACK options . So as Tx3 too。
我用的XM+  sbus接收机接在rx2，图传控制线接在tx1，tx1的选项选择了TBS SmartAudio。使用X9D+遥控器进入OSD的菜单》VTX SA，菜单里面VTX SMARTAUDIO并没有显示当前图传频率信息，只有虚线，和 STAT、BACK 两个选项。换到tx3也是一样的。

3.) Some one said that board's serial port (tx1 tx3) inverters were opened when SBUS reciver was selected in rx2。
有人说由于接收机接在rx2，选择SBUS打开了所有串口反相器，导致Tx1 Tx3也反相了。

DooDoo512, 2017-09-04:
Thank you for your comments…

jflyper, 2017-09-04:
@DooDoo512 Your problem is a board specific configuration problem. Please redirect your support request to appropriate support arena on the net, such as RCG. You may also contact your distributor.

Thanks in advance.

EDIT
FYI, from the target definition file for BEEROTORF4 target (I assume this is your board), UART2 and UART3 seems to be equipped with inverters, and these are not compatible with SmartAudio.
This leaves UART1 (TX1) for your SmartAudio connection, and if it does not work, you have to go for software serial.

closed 2017-09-04 by jflyper

DooDoo512, 2017-09-04:
Thanks ! @jflyper 

I have tested UART1(TX1) , it's the same situation as UART3 .
And this board has no software serial port from it‘s Manufacturer, TOT

mikeller, 2017-09-04:
@jflyper: 

> ...these are not compatible with SmartAudio...

What is the story of switchable inverters not being compatible with SmartAudio?

jflyper, 2017-09-05:
@mikeller 
SmartAudio and Tramp driver is not capable of handling full-duplex.

mikeller, 2017-09-05:
Ah, so what we need is a way to set `half_duplex` for the VTX control serial port?

jflyper, 2017-09-05:
Yeah, `set half_duplex = OFF`.

mikeller, 2017-09-05:
Shouldn't be too hard, and will be helpful for a lot of the F4 targets. Question is do we want to do this as a stopgap measure for 3.2?

DooDoo512, 2017-09-05:
Very very appreciated !

So , can I set `set half_duplex = OFF`  in CLI ?
I'm willing to test it

mikeller, 2017-09-05:
No, not yet, the parameter will have to be added for the VTX control UART.

DooDoo512, 2017-09-05:
OK ,  looking forward to  see that 
O(∩_∩)O 哈哈~

mikeller, 2017-09-05:
@DooDoo512: Try this:

[betaflight_3.2.0_BEEROTORF4.zip](https://github.com/betaflight/betaflight/files/1276770/betaflight_3.2.0_BEEROTORF4.zip)

(The setting is `vtx_halfduplex`, it needs to be `off`, and this will only work on UART3, SmartAudio connected to RX, with a diode between TX and RX, cathode side towards RX.)

DooDoo512, 2017-09-08:
@mikeller 
emmmmm~  no change, the problem still exists

mikeller, 2017-09-11:
@DooDoo512: Can you draw a diagram (or post a picture) of your wiring?

DooDoo512, 2017-09-11:
Is this right ?
[image]

mikeller, 2017-09-11:
No, the diode should be the other way round (i.e. the opposite direction of how it's done for SmartPort, since SmartPort uses inverted serial, and SmartAudio doesn't).

DooDoo512, 2017-09-11:
Then , anode towards RX .
OK I'll try

mikeller, 2017-09-11:
Yes. My bad. Amended comment.

DooDoo512, 2017-09-18:
@mikeller  Sorry for late , I have tested , did not work...

mikeller, 2017-09-18:
:-(

jflyper, 2017-09-18:
@DooDoo512 
I think you have to scope the signals.

DooDoo512, 2017-09-18:
@jflyper  Yes, U R right .However ,thank you so much !

PR #4066 (2017-09-07): Added halfduplex option for VTX control ports.
Fixes #4059.