betaflight/betaflight #4062: Compiling errors

Tomcatt, 2017-09-04:
Hello,

Was compiling different build to tinker with and I'm getting the following errors:

Building SPRACINGF3
make[2]: mkdir: Resource temporarily unavailable
/bin/sh: fork: Resource temporarily unavailable
ccache: error: Failed to fork: Resource temporarily unavailable
make[2]: *** [obj/main/SPRACINGF3/sensors/sonar.o] Error 128
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [obj/main/SPRACINGF3/io/osd.o] Error 1
make[2]: *** wait: No child processes.  Stop.
make[1]: *** [binary] Error 2
make: *** [SPRACINGF3] Error 2

Building REVO
/bin/sh: fork: Resource temporarily unavailable
ccache: error: Failed to fork: Resource temporarily unavailable
ccache: error: Failed to fork: Resource temporarily unavailable
make[2]: *** [obj/main/REVO/drivers/display_ug2864hsweg01.o] Error 128
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [obj/main/REVO/common/gps_conversion.o] Error 1
make[2]: *** [obj/main/REVO/common/colorconversion.o] Error 1
ccache: error: Failed to fork: Resource temporarily unavailable
make[2]: *** [obj/main/REVO/fc/fc_init.o] Error 1
make[1]: *** [binary] Error 2
make: *** [REVO] Error 2

Building REVO_OPBL
ccache: error: Failed to fork: Resource temporarily unavailable
make[2]: mkdir: Resource temporarily unavailable
make[2]: mkdir: Resource temporarily unavailable
make[2]: *** [obj/main/REVO_OPBL/common/colorconversion.o] Error 1
make[2]: *** Waiting for unfinished jobs....
ccache: error: Failed to fork: Resource temporarily unavailable
make[2]: *** [obj/main/REVO_OPBL/common/gps_conversion.o] Error 1
ccache: error: Failed to fork: Resource temporarily unavailable
make[2]: *** [obj/main/REVO_OPBL/drivers/display_ug2864hsweg01.o] Error 1
make[2]: *** wait: No child processes.  Stop.
make[1]: *** [binary] Error 2
make: *** [REVO_OPBL] Error 2

Building SPARKY2
/bin/sh: fork: Resource temporarily unavailable
ccache: error: Failed to fork: Resource temporarily unavailable
make[2]: *** [obj/main/SPARKY2/drivers/sonar_hcsr04.o] Error 128
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [obj/main/SPARKY2/drivers/vtx_common.o] Error 1
make[1]: *** [binary] Error 2
make: *** [SPARKY2] Error 2

Building REVOLT
make[2]: mkdir: Resource temporarily unavailable
make[2]: *** [obj/main/REVOLT/common/gps_conversion.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [binary] Error 2
make: *** [REVOLT] Error 2

I'm not a strong coder but if it's missing library files I'm good at finding / installing those

Arakon, 2017-09-04:
I can compile just fine in cygwin under windows 7.. using a normal pulled copy of this git. 
Linking SPRACINGF3
arm-none-eabi-size ./obj/main/betaflight_SPRACINGF3.elf
   text    data     bss     dec     hex filename
 237377    9080   28032  274489   43039 ./obj/main/betaflight_SPRACINGF3.elf
arm-none-eabi-objcopy -O ihex --set-start 0x8000000 obj/main/betaflight_SPRACING
F3.elf obj/betaflight_3.2.0_SPRACINGF3.hex
make[1]: Leaving directory '/cygdrive/d/betaflight'

Doubt this has anything to do with the code, rather your environment.

Tomcatt, 2017-09-04:
Ah, that sucks.  Okay, how do I check this?  I use a Mac for all compiling in the past.  (What I get for leaving the game for a few months) LOL

Tomcatt, 2017-09-04:
Okay the fix I found was just reinstall "re-brew" my arm files.  Recompiled and all is well.  @Arakon thanks for the input bro!

closed 2017-09-04 by Tomcatt