betaflight/betaflight #4063: i feel PREARM is rather TWOCONDITIONARM

piepluquad, 2017-09-04:
Hi folks,

I have a X9D+ taranis in my setup.
I use SF switch for ARM via AUX, and SH switch for PREARM via AUX

If i first activate SF in UP position (ARM is engaged), then a single UP tap on my SF switch will arm my quad...
so I don't feel like it's a 'PREARM' but rather a 'TWO CONDITION ARM'

if it was really PREARM, i couldn't set the ARM ON if PREAM is not ON before
a real PREARM would  oblige user to trigger PREARM first, and while PREARM is on permits ARM switch

do I have a bug or misunderstanding on my side ?

swilkens, 2017-09-04:
I wonder if that is because you are on RC2, I believe this was fixed later. See this pull request:

https://github.com/betaflight/betaflight/pull/3738

piepluquad, 2017-09-04:
ok, i'll close this issue and 'll see when i update my quad, thanks @swilkens

closed 2017-09-04 by piepluquad