betaflight/betaflight #4065: [Question] motor_pwm_rate correct ?

robert-hoffmann, 2017-09-05:
BF 3.2 RC3

I just noticed that my **motor_pwm_rate** is set at **480** while running DSHOT600 (BF defaults apparenntly), is this the correct value ?

I'm asking because the documentation says
> set unsynced_fast_pwm = OFF
[ON,OFF]
unsynced motor update speeds for fast PWM protocols up to 32k with **Set motor_pwm_rate = [0-32000] (Oneshot125 not over 4000, os42 not over 12000, ms right up to 32k)**

I'm asking because i just set the PWM frequency of the ESC's in blhelisuite to 48khz ...should i be touching **motor_pwm_rate** to make it match or something like that ..or is it even supported that high... or does this only apply if using **unsynced_fast_pwm** ..and if so, do you think in my case i should be using this ?

Any insight would be appreciated,
Thanks !

DieHertz, 2017-09-05:
PWM rate of ESC is the frequency of signal driving motor phases, it has nothing to do with motor rate in BetaFlight. You're good.

robert-hoffmann, 2017-09-05:
Good to know, thanks !

closed 2017-09-05 by robert-hoffmann