betaflight/betaflight #4067: SPRacing F3 EVO - Voltage sensing not working

swilkens, 2017-09-05:
**Hardware:**
SPRacing F3 EVO

**Software:**
Betaflight 3.2.0-RC4
Betaflight Configurator 3.2.1 (Chrome Store)

**Set-UP:**
Quadcopter X using DSHOT 600 with remapped motor 4 according to the wiki "Known Issues" for this board.
https://github.com/betaflight/betaflight/wiki/Board---SPRACINGF3EVO

**Situation description:**
Quad is on the bench, connected over USB to a pc running windows 10 . Configurator connected. LiPO Attached. Voltage sensor is set to on-board ADC.

**Problem Description:**
After having properly connected the battery + and - to the SPRacing F3 EVO's battery headers (as per the manual [1]), I am unable to see a current battery voltage. 

**Troubleshooting attempted:**
I have manually verified that the battery headers are in fact properly connected. Using a multi-meter, I have verified that the battery voltage is in fact available on the pins.

I have not verified using Cleanflight.

Attached a full dump and a diff all.

@hydra ?

[fulldump.txt](https://github.com/betaflight/betaflight/files/1278891/fulldump.txt)

[diffall.txt](https://github.com/betaflight/betaflight/files/1278892/diffall.txt)

jflyper, 2017-09-06:
@swilkens You are using RC3. Try RC4.

richard-scott, 2017-09-06:
@swilkens Is this an original board, or a clone? I have an original board and it is working fine for me (I was using hex files from ci.betaflight.tech tho which may be where I'm doing it differently). I can double check tonight to see if I get issues with RC3 and RC4 etc.

Did you connect vbat to the pins identified by the pink box bottom right?

[image]

swilkens, 2017-09-06:
@jflyper oops, too many quads - seems I forgot to flash this one. Either way, problem also exists on RC4.

@richard-scott this is an original board, vbat is properly connected to the pink battery headers on the bottom **left**. (checked by measuring battery voltage on the pins). Which firmware version are you running?

swilkens, 2017-09-06:
Since other users @ rc-groups and facebook have confirmed this should work, I am concluding this to be a hardware issue with my board.

closed 2017-09-06 by swilkens

swilkens, 2017-09-06:
Confirmed hardware failure, was able to restore.

[image]

jflyper, 2017-09-07:
This happens with the authentic board?

swilkens, 2017-09-07:
@jflyper yes, this is on an authentic board. I'm not sure what caused this damage, it seems like a cold break to me but the left side of the SMD was still attached.

Either way - I have restored the SMD and voltage sensing is now working again.

mikeller, 2017-09-07:
Could be a manufacturing defect - even with authentic boards I doubt they can do much QA, considering the prices they are sold at. Probably just made contact by pressure for some time.