betaflight/betaflight #4068: Beeper -ON_USB doesn't disable SYSTEM_INIT beeps

swilkens, 2017-09-05:
**Hardware:**
SPRacing F3 EVO
Betaflight F3

**Software:**
Betaflight 3.2.0-RC4, -RC3
Betaflight Configurator 3.2.1 (Chrome Store)

**Problem Description:**
Setting -ON_USB to disable beeps while the quad is on the bench does not disable the SYSTEM_INIT beep. Subsequently setting -SYSTEM_INIT does disable the beeping sequence. 

Is this intended behavior and is the SYSTEM_INIT excluded from -ON_USB?

So:

Just -ON_USB: SYSTEM_INIT beeps still heard while on bench
Both -ON_USB and -SYSTEM_INIT: no beeps heard while on benc

mikeller, 2017-09-05:
`SYSTEM_INIT` isn't specifically excluded from `ON_USB`, but the system init beeps happen before USB detection, so the firmware does not yet have the information that it is connected on USB.

swilkens, 2017-09-06:
That makes sense, I'll update the wiki page and reference this issue. Thank you @mikeller 

Closing this.

closed 2017-09-06 by swilkens