betaflight/betaflight #4070: F405 32khz + dynamic filter = desync

darksonny, 2017-09-06:
///
I've been trying to get 32/16khz combination running, but my motors get noisy and hot after seconds.

My setup, BF 3.2 RC4,Matek F405 softmount, DYS XSD v2, Emax RS2205S.
...
Case 8/8khz:
Dynamic Filter, anti gravity, air mode, OSD - ON.
Filter settings PT1, no notch filter active.
No SDcard logging, everything else default.
Dshot600, motor idle 5
works perfectly.
...
Case 32/16khz or 16/8khz (32khz sampling enabled)
As soon as I start to change from BIQUAD to PT1 or even remove the filters, the motors gets hot and noisy, start to desync.
The weird thing, I changed back to 8/8khz in 32khz sampling mode, still the same effects.

martinbudden, 2017-09-06:
The ICM206xx series gyros are much noisier when run with 32k sampling enabled, and so require more filtering when run in this mode.

Closing, since this is know ICM206xx series gyro behaviour.

closed 2017-09-06 by martinbudden

darksonny, 2017-09-06:
but what about I flip the 32khz switch, but set it to 8/8khz?
In that case, it's still noisy as hell. Why?

DieHertz, 2017-09-06:
That's what martin said, 32K sampling is noisy on the ICM206xx series. If you flip the 32KHz switch, you still get 32 KHz sampling decimated to 8KHz.

darksonny, 2017-09-06:
ahhh ok, thanks for the fast answers here. ;)