betaflight/betaflight #4072: feature request: Smart audio power on switch

supiiik, 2017-09-06:
Hi, would it be possieble to implement new functionality for SmartAudio? I'd like to assign power change for VTX by the transmitter switch. We are usually flying in group of 3 or 4 persons, so we have to be at 25mW, but sometimes happens, that the signal is weak, when I fly behind the wall, of far away. Could be great to have an option to change the VTX power by the switch on the transmitter and avoid video signal loss in critical situations. Thanks

Jahnkeanater, 2017-09-09:
If this becomes a feature It should also be added to the black box log. I could see this being abused during a race. When flying in a group it's generally a bad idea for anyone to be on a higher power setting. If you are flying with only 4 people you should be able to use 200mw without issues. At races, we have had 6 pilots at a time on 200mw without issues. We switched to 25mw this year for 8 pilots.

mikeller, 2018-02-11:
Duplicate of #3094.

closed 2018-02-11 by mikeller