betaflight/betaflight #4075: [Feature Request] - limit gyro loop time on F3 board which utilize dynamic filter

haybeseret, 2017-09-07:
Hi,
As per release notes for dynamic filter and per mjbudden comment on Boris 
blog ([here](https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread/page3430#post38212022))

I think it will be wise to prevent users with F3 boards to set the gyro loop time to 8k.

Thanks,

Arakon, 2017-09-07:
Why?
I for one see absolutely zero negative effect from running 8k with dynamic filters on. I think this should be the user's decision whether they want to 'risk' it or not.

martinbudden, 2017-09-07:
I'm with @Arakon here, even though it was my recommendation. One of the ways betaflight advances is by pilots doing things things they aren't suppose to do. I still don't recommend running 8K with dynamic filters on F3 boards, but if people want to do that, I don't think we should stop them. Pilots should make their own risk/benefit choices.

closed 2017-09-07 by martinbudden

swilkens, 2017-09-07:
This seems like the sort of thing the configurator could warn a user for. A small "?" next to the gyro rate could just mention that 8k on F3 boards (when the board is detected as such) is an edge case likely to overload the CPU during flight with DF enabled.

Or perhaps a better place would be the "?" next to the Dynamic Filter enable switch in the configurator.

haybeseret, 2017-09-08:
I am ok with whatever you will decide, 
It's just not aligned with the release notes of BF3.2 [here](https://github.com/betaflight/betaflight/wiki/BetaFlight-3.2#dynamic-filters)
. so either amend the release notes or deliver this validation to the configurator.

mikeller, 2017-09-11:
@haybeseret: I think they are aligned. The 'don't use 8k on F3' is a recommendation, not a hard limitation. In Betaflight, we generally try to prevent configurations that will clearly not work (e.g. running OneShot125 at >= 4k), but we allow users to experiment with things that are a bit experimental. 8k on F3 might well work if everything else is switched off...

swilkens, 2017-09-11:
So for the sake of being complete, I created a bare minimum setup on an F3 chip running 3.2.0-RC4 and evaluated the performance. I should note that, while it is evident that the mean result is below what would be considered optimal, the quad still flies very stable. All other filters and features are disabled, including logging.

n= 50 

We can see that a bare minimum setup on 8/8 with dynamic filtering on more almost tipples the deviation from the expected gyro rate compared to a 4/4 situation with dynamic filtering. Perhaps more importantly, while the highs aren't affected so much, the minimums certainly are.

Based on this, I would think it's useful to at least note this in the configurator. Considering this is an absolute minimal setup.

[image]

haybeseret, 2017-09-11:
@swilkens. Thanks for the testing!
I agree that there should be a note. Its a very good suggestion.

martinbudden, 2017-09-11:
@swilkens , thanks for testing.

I'm not sure what you mean by "deviation" however. For 8k/8k I make deviation to be (7936-6134)/7447 , that is approx 24%.

swilkens, 2017-09-11:
Sorry if the term implies something else, English isn't my native language. Deviation is intended to be a measure of how far the averaged GYRO rate is off from the ideally expected rate. I.E. percentually, how far is the averaged GYRO rate off, assuming 8000 (or 4000 in the case of 4k) is 0% off.

For 8k/8k that would be 7447/8000 = 93.1%, or a deviation of 6.9%. Rounded off obviously.

It'll be interesting to run all of this with a higher n and make some box plots with standard deviations to get a better picture of the impact of the various features on the actual performance.  I plan to do some more testing with enabling / disabling features over the next week, hopefully to map the performance impact (and spread of the performance impact) on a per feature basis.

martinbudden, 2017-09-11:
OK, I understand what you mean. However in terms of quadcopter performance the deviation, if consistent, is not particularly harmful. If at 8k/8k the gyro rate was consistently (say) 7100, then that would not matter too much - it would just mean the tune would be slightly different.

What's much more harmful is the 'jitter' - the variation in rate. So the fact that the gyro rate can vary from 6134 to 7936 (ie about 24%) is not good at all. Betaflight is reasonably resistant to jitter, however it is still something that should be avoided.

swilkens, 2017-09-11:
The thinking was that a higher deviation could indicate less headroom in the CPU, making it less capable of dealing with the jitter. But I see your point, it makes sense. As said, I'll do some more extensive testing over the week, I'll add the jitter (and spread thereof).

swilkens, 2017-09-11:
@martinbudden I worked out the max jitter for the datasets I gathered before:

[image]

Now just to clarify:
GYRO RATE lists the Minimum, Maximum and averaged (50 samples) GYRO RATE values.
GYRO JITTER lists the Minimum, Maximum deviation from the average GYRO RATE and additionally shows the absolute difference between Min and Max. Diff % shows that same difference as a percentage of the average GYRO RATE.

Note that this is with all other filters off and features off, it really shows the dramatic impact of the Dynamic Filter of the level of jitter, as you call it.