betaflight/betaflight #4080: Omnibus f4 pro corner and ominibusF4SD target

mStrangers, 2017-09-07:
On the omnibus f4 pro corner there 2 inverter connected to PC09 for UART3 and PC08 for UART6
in the actual nightly an RC3 RC4 ,only the inveter for UART6 Is declared , inverter for UART3 is declared as PWM6 so the port for SBUS ( as discribed in the pinout) didn't work or work parcialy

jflyper, 2017-09-07:
@mStrangers 

https://github.com/betaflight/betaflight/wiki/Board---OMNIBUS-F4-Pro-Corners

closed 2017-09-07 by jflyper

jflyper, 2017-09-07:
@mStrangers 

As use case of CH6 (PWM6) are limited and non-standard, I will consider making PC9 as inverter control by default.

mStrangers, 2017-09-07:
sorry i've searched in issue not in wiki i'm apologies.
The change in the next RC is a good thing i think.

jflyper, 2017-09-08:
#4092