betaflight/betaflight #4086: BF 3.2RC4 - Can't download blackbox logs on Naze32 rev6

etracer65, 2017-09-07:
The configurator progress bar appears but remains empty and doesn't change.  Have to quit and restart the configurator and can then reconnect without rebooting the board so it doesn't look like it crashed or is hung.  Appears to be some incompatibility between the configurator and BF 3.2RC4 affecting the NAZE target (or maybe all F1's).

Reflashing BF 3.1.7 restores the blackbox log download capability.

Tried the latest configurator from the repo and still fails.  Looked at the javascript console and no errors are reported.

mikeller, 2017-09-08:
Fixed by #4088.

mikeller, 2017-09-08:
@etracer65: Can you check with latest from master?

etracer65, 2017-09-08:
@mikeller   PR4088 fixed the problem.  I was able to download logs from a Naze32 rev6 (NAZE target) without any problems.

closed 2017-09-08 by mikeller